ct-Open-Source/tuya-convert #759: Blitzwolf BW-SHP8 smart socket problem

andrkaz, 2020-09-05:
Can not flash this socket via Tuya-convert, error is "decryption failed". I flash this device by wire and backup original formware, may be this help to improve Tuya-convert? https://drive.google.com/file/d/1xacWfFQHPvsd8I7vlY8OHG7xyuFz2yNh/view?usp=sharing

gisab, 2020-09-05:
The same for me for another kind of device. 
I have photo, firmware and procedure availabe for sharing. 
Please send me instructions on how to share the material. 

Inviato da iPhone

KTibow, 2020-09-06:
@gisab just drag and drop it into the comment area.

codedmind, 2020-09-26:
Any news regarding this issue?

Brand new shp8 cannot be converted

schilea, 2020-11-17:
Same for me. Flashed before 4 SHP8 devices without problem but the latest one bought in August is not working.

declanshanaghy, 2020-11-17:
I ran into this on a Gosund WP3 last night. It's the same issue as https://github.com/ct-Open-Source/tuya-convert/issues/483

Work is ongoing in this wiki now: https://github.com/ct-Open-Source/tuya-convert/wiki/Collaboration-document-for-PSK-Identity-02