ct-Open-Source/tuya-convert #760: rpi 3b+ not visible access point vtrust-flash after start script start_flash.sh

Nikovr4, 2020-09-09:
Good day, im flashing new clean image on sd card, and  create 3 command

> cd /tmp
> git clone https://github.com/ct-Open-Source/tuya-convert.git
> cd tuya-convert

after this, run ./start_flash.sh
[[image]](url)
next step i need find my phone access point with name **vtrust-flash**. but on this step a can't find this SSID
this is my console out of command ifconfig, you can see wlan0 is active and enable
[[image]](url)
Please, tell me what i need do?
if i start script **tmp/tuya-convert/scripts/setup_ap.sh**, then access point  with name **vtrust-flash**  is can find, but without DHCP server, after i can connect to **vtrust-flash** with  my  phone, i cant receive IP address

chrisp42, 2020-09-10:
As per the installation instructions: https://github.com/ct-Open-Source/tuya-convert#installation
I would guess you didn't install the prerequisites.
`# ./install_prereq.sh`
If you have installed the prerequisites, we will need more logs to determine what the problem is.

drumstick77, 2020-12-01:
Hi, I did install the prereq.sh and now I'm getting this issue.

anyone seen this before? I flashed 8 of these plugs 1.5 years ago with no issues
[image]

craigrdaniels, 2024-05-27:
I was just having the same issue. Had to do the following command before running the script and it appears to have worked (new tuya firmware so OTA failed anyway).

sudo systemctl stop NetworkManager