ct-Open-Source/tuya-convert #763: Curtain Switch reacts but reboots immediately

CrazyChris75, 2020-09-13:
Hi Tuya and Tasmota Gurus,

I am trying to flash these switches: https://templates.blakadder.com/SCS86-03AJAI.html
using docker, tuya-convert works: I can connect my phone to the AP!

But when I hold the down-button for 20 seconds to put the curtain switch into pairing mode
it flashes the LED just once or twice and then seems to "reboot" immediately
and tuya-convert shows

...................
SmartConfig complete
Resending SmartConfig

10 times and then gives up

The Smart Life app shows firmware and MCU version 1.0.7 

Does this mean that the curtain switch refuses OTA and simply reboots immediately?
Is that firmware version not flashable any more?
Is there anything else I can try to get my switches out of the cloud?

Thank you!

kueblc, 2020-09-13:
Please include your logs when opening an issue

CrazyChris75, 2020-09-14:
> Please include your logs when opening an issue

Sorry, it took me a while to get this done since I think the log output is not much help here.

Here is a video where you can see how the switch immediately reboots/resets when tuyaconvert starts pairing:
https://youtu.be/7_jgGWeKUCk

And here is the log output:

```
sudo docker-compose exec tuya start
tuya-convert v2.4.4
======================================================
TUYA-CONVERT

https://github.com/ct-Open-Source/tuya-convert
TUYA-CONVERT was developed by Michael Steigerwald from the IT security company VTRUST (https://www.vtrust.de/) in collaboration with the techjournalists Merlin Schumacher, Pina Merkert, Andrijan Moecker and Jan Mahn at c't Magazine. (https://www.ct.de/)

======================================================
PLEASE READ THIS CAREFULLY!
======================================================
TUYA-CONVERT creates a fake update server environment for ESP8266/85 based tuya devices. It enables you to backup your devices firmware and upload an alternative one (e.g. ESPEasy, Tasmota, Espurna) without the need to open the device and solder a serial connection (OTA, Over-the-air).
Please make sure that you understand the consequences of flashing an alternative firmware, since you might lose functionality!

Flashing an alternative firmware can cause unexpected device behavior and/or render the device unusable. Be aware that you do use this software at YOUR OWN RISK! Please acknowledge that VTRUST and c't Magazine (or Heise Medien GmbH & Co. KG) CAN NOT be held accountable for ANY DAMAGE or LOSS OF FUNCTIONALITY by typing yes + Enter

yes
Checking for network interface wlo1... Found.
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen..................................
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
...........
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N] 

======================================================
Cleaning up...
No screen session found.
Closing AP
Exiting...
```

BTW there are no *.log files anywhere after this.

Cheers, Chris

finnzz, 2020-09-14:
Log files are in the scripts folder of your TuyaConvert directory

CrazyChris75, 2020-09-14:
alrighty, managed to get the files out of the docker container
(wow ... maybe I do need to get a Raspberry Pi after all *facepalm*)

[tuya-convert.zip](https://github.com/ct-Open-Source/tuya-convert/files/5219242/tuya-convert.zip)

kueblc, 2020-09-14:
Unfortunately you've been hit by #483. Subscribe for updates or follow collaboration on the wiki.

closed 2020-09-14 by kueblc

CrazyChris75, 2020-09-14:
damn it!

hope this can be solved - anyways, thank you very much!