ct-Open-Source/tuya-convert #765: Lerlink X802A-RF Flash tasmota failed. Lerlink changed the Tuya chip from TYWE3S to WB2S.

horacecar, 2020-09-16:
**Did you have issues getting a device to flash?**
Lerlink X802A-RF Flash tasmota failed using the latest tuya-convert v2.4.4.

Half year ago they are using the left Tuya chip TYWE3S, now they changed to use Tuya new chip WB2S. (REF. Picture A). 
The log i uploaded below. thank you

Please Help 👍 

**log**
---------------smarthack-mqtt.log Start------------------------------------

^C1595359186: mosquitto version 1.4.15 terminating
1595359306: mosquitto version 1.4.15 (build date Tue, 18 Jun 2019 11:42:22 -0300) starting
1595359306: Using default config.
1595359306: Opening ipv4 listen socket on port 1883.
1595359306: Opening ipv6 listen socket on port 1883.
^C1595369807: mosquitto version 1.4.15 terminating
1600098961: mosquitto version 1.4.15 (build date Tue, 18 Jun 2019 11:42:22 -0300) starting
1600098961: Using default config.
1600098961: Opening ipv4 listen socket on port 1883.
1600098961: Opening ipv6 listen socket on port 1883.
^C1600099191: mosquitto version 1.4.15 terminating
1600099320: mosquitto version 1.4.15 (build date Tue, 18 Jun 2019 11:42:22 -0300) starting
1600099320: Using default config.
1600099320: Opening ipv4 listen socket on port 1883.
1600099320: Opening ipv6 listen socket on port 1883.
---------------------------end---------------------------------

could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:852)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.23:50893
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:852)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.23:50894
could not establish sslpsk socket: [SSL: WRONG_SSL_VERSION] wrong ssl version (_ssl.c:852)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.23:50895
could not establish sslpsk socket: [SSL: WRONG_SSL_VERSION] wrong ssl version (_ssl.c:852)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.23:50896
could not establish sslpsk socket: [SSL: WRONG_SSL_VERSION] wrong ssl version (_ssl.c:852)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.23:50897
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:852)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.23:50898
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:852)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.23:50899
could not establish sslpsk socket: [SSL: WRONG_SSL_VERSION] wrong ssl version (_ssl.c:852)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.23:50900
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:852)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.23:50901
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:852)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.23:50902
could not establish sslpsk socket: [SSL: WRONG_SSL_VERSION] wrong ssl version (_ssl.c:852)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.23:50903
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:852)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.23:50904
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:852)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.23:50905
could not establish sslpsk socket: [SSL: WRONG_SSL_VERSION] wrong ssl version (_ssl.c:852)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.23:50906
[... 434 lines cut ...]
wlx00e04c817803: AP-STA-CONNECTED c0:a6:00:a3:9a:ca
wlx00e04c817803: AP-STA-DISCONNECTED c0
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5231328/smarthack-mqtt.log)
:a6:00:a3:9a:ca
wlx00e04c817803: AP-STA-CONNECTED c0:a6:00:a3:9a:ca
wlx00e04c817803: AP-STA-DISCONNECTED c0:a6:00:a3:9a:ca
wlx00e04c817803: AP-STA-CONNECTED c0:a6:00:a3:9a:ca
wlx00e04c817803: AP-STA-DISCONNECTED c0:a6:00:a3:9a:ca

------------------end --------------------

file upload
photo:
[image]
[image]
[image]

log:
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/5231334/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5231335/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5231336/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5231337/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/5231339/smarthack-udp.log)

**Please remember to close your issue when the problem has been addressed!**

kueblc, 2020-09-16:
tuya-convert and Tasmota only support the ESP82xx chipset.

closed 2020-09-16 by kueblc

he-so, 2021-04-02:
@horacecar you can replace the WB2S chip by a TYWE2S, that is 1:1 pin compatible. There is one shop selling them on ali*ess. I replaced them in two power monitoring sockets to make tasmota work. Sure, knowing it in advance, i'd order something else. But this fixes it for less.