ct-Open-Source/tuya-convert #766: FEIT Smart Dimmer - Failing to Flash

MxD-js, 2020-09-17:
I bought (2) FEIT Electric Smart Dimmer back earlier in the year, just got around to opening the package. Flashed successfully with tuya-convert, then bought 3 additional dimmers 6 months down the road, and cannot flash. Get the dreaded in the smarthack-psk.log

`could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac`

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5242497/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5242498/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/5242499/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/5242500/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5242501/smarthack-wifi.log)

Any idea what could be the issue? 
I did do some digging and see that tuya patched the vuln that was used previously, but am not sure.

kueblc, 2020-09-17:
> I did do some digging and see that released patches and the dimmer may have been patched

You are correct, this device is blocked by #483. No solution in the foreseeable future but subscribe for updates and/or follow along in the wiki in case something changes.

closed 2020-09-17 by kueblc

MxD-js, 2020-09-18:
Thanks. Since it is not possible OTA. Has anyone attempted to flash the chip via serial? Seems like the device has the TYWE2S chip which in turn is a ESP8285, and there are guides online on flashing it for other smart devices, but the chip is soldered on upside down, meaning GPIO 0 is on the underside of the PCB. You'd have to de-solder the TYWE2S chip before flashing. 

Not the best path forward, but is a path none the less.

BlowOnFuel, 2020-10-04:
I got few of those which also could not be flashed OTA with TuyaConvert.

I took off the TYWE2S with a heat gun and was successful at flashing the TYWE2S and reinstalling them after reloading the contacts. As there were no apparent traces/contacts under the TYWE2S, I tried poking a hole to gain access to GPIO0 but it still does not seem to enable flashing the ESP with the PCB is mounted on the mainboard. I suspect the TuyaMCU on TX/RX also prevents access to the lines for flashing.

Hope that helps.