ct-Open-Source/tuya-convert #767: Flashing of Sonoff BASICR2 WiFi Switch with a Raspberry Pi

gisab, 2020-09-18:
Device: https://sonoff.tech/product/wifi-diy-smart-switches/basicr2

Precondition: Configure Raspberry PI inverting AMA and S0 serial in config.ini and cmdline.txt
https://spellfoundry.com/2016/05/29/configuring-gpio-serial-port-raspbian-jessie-including-pi-3-4/
https://hallard.me/enable-serial-port-on-raspberry-pi/

Connection:
# connect RPI GPIO 5V to pin 3 of 5V/3V converted
# connect RPI GPIO TX to ESP serial RX
# connect RPI GPIO RX to ESP serial TX
# connect RPI GPIO GND to ESP GND

Steps:
# switch on the device with ESP GPIO0 shortcut on GND; this can be done by pressing the device button
# run esptool and flash the firmware

{code}
esptool.py erase_flash
esptool.py write_flash -fm dout 0x0 tasmota-IT.bin
{code}

kueblc, 2020-09-18:
tuya-convert has no relation to Sonoff

closed 2020-09-18 by kueblc

gisab, 2020-09-18:
Hi Colin.
You told me in the past to create an issue and to upload the phono in the
comment section.
I was trying to follow your procedure.
If you believe the photo and the procedure are useful to be shared, just
tell me.

(you closed the issue before I was able to complete it)

Ciao,
Gianluca

gisab, 2020-09-18:
Hi kueblc.
Someone told me in the past to create an issue and to upload the photos in the
comment section.
I was trying to follow your procedure.
If you believe the photo and the procedure are useful to be shared, just
tell me.

kueblc, 2020-09-18:
You can use the wiki to share information you think might help others in the community, I see you already have a page started. I use issues to track problems people are having with the tuya-convert software.

gisab, 2020-09-18:
(facepalm)
Actually i deleted as I was creating the Issue.
In the wiki I don't know how to upload images.
It seems that it's only possible to link internet images.

kueblc, 2020-09-18:
OK I understand now, sure you can use this closed issue to upload your images and link them on your wiki page.

kueblc, 2020-09-18:
Given that tuya-convert has no relation to Sonoff, I might look at Tasmota or another relevant community for the best place for this information. I do not want users to be confused by your article into thinking that tuya-convert can flash Sonoff devices.

gisab, 2020-09-19:
[image]
[image]
[image]
[image]
[image]

gisab, 2020-09-20:
[image]