ct-Open-Source/tuya-convert #768: Lerlink X802A-RF Flash tasmota failed. I brought a new wall switch today with TYWE3S Chip.

horacecar, 2020-09-18:
### Before opening a new issue, please read the following

**Did you have issues getting a device to flash?**

Lerlink X802A-RF Flash tasmota failed using the latest tuya-convert v2.4.4.

 I brought a new wall switch with TYWE3S chip today, but still cannot flash with tuya-convert . (REF. Picture A).
The log i uploaded below. thank you

[image]
[image]
[image]

[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5247407/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/5247409/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/5247410/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5247412/smarthack-wifi.log)

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5247406/smarthack-mqtt.log)

Please Help 👍
**Please remember to close your issue when the problem has been addressed!**

kueblc, 2020-09-18:
Thank you for filling in the issue template and providing all the requested details.

Unfortunately you've hit #483. Tuya has patched the vulnerabilities we use to spoof the device onto our fake cloud. No solution in the foreseeable future but subscribe for updates and/or follow along in the wiki in case something changes.

closed 2020-09-18 by kueblc

horacecar, 2020-09-19:
@kueblc  hello. Thanks for the reply, can I use the wire hack (esptools) to flash the Firmware for the latest tywe3s tuya chip?

kueblc, 2020-09-19:
Yes you should be able to flash this by wire

vizras, 2021-09-23:
What contacts needs? Please give scrin.