ct-Open-Source/tuya-convert #769: Unable to flash BN-Link BNC-60/U133TJ

unmesh59, 2020-09-26:
Not sure how to get the firmware version but these are new out of the box and I have used my Raspberry Pi setup to successfully flash other Tuya based smart plugs

The plug is in fast blink mode, but the LED turns off soon after the Pi starts sending SmartConfig packets, then turns to a steady blue after a few seconds while the tuya_convert script keeps retyring before giving up.

Maybe these have the patched OEM firmware but I thought I'd ask here before returning the plugs.

Logs attached.

P.S.: I could buy TopGreener TGWF115PQM but couldn't find tasmota templates for them

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5286019/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5286020/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/5286021/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/5286022/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5286023/smarthack-wifi.log)

kueblc, 2020-09-26:
Unfortunately you are correct, these have been patched. See #483 and the wiki for more information.

closed 2020-09-26 by kueblc

unmesh59, 2020-09-26:
Thank you for the link to the wiki entry. The TopGreener devices are not on it so I will return the BN-Links, order those and hope that I can find the template info somewhere