ct-Open-Source/tuya-convert #770: AP mode not supported! TP-LINK TL-wn725

sanderlv, 2020-09-26:
Hello,

installed both kali and ubuntu on pi3.

Both connected with a TPlink TE7WN725N wifi adapter. all running fine but I get:
```
ubuntu@ubuntu:~/tuya-convert$ ./start_flash.sh
tuya-convert v2.4.4
AP mode not supported!
Please attach a WiFi card that supports AP mode.
======================================================
Cleaning up...
No screen session found.
No screen session found.
No screen session found.
No screen session found.
Closing AP
Exiting...

```
What can be wrong?

sanderlv, 2020-09-26:
Not sure why it is not working. Should work right?

I also ordered according to http://www.bernaerts-nicolas.fr/iot/363-tasmota-firmware-ota-tuya-switch:
https://www.amazon.de/dp/B002PD61Y4/ref=cm_sw_em_r_mt_dp_245BFbTFWPE8E

kueblc, 2020-09-26:
I found this article which may help you:
https://raspberrypi.stackexchange.com/questions/22259/ap-with-tl-wn725n-v2

sanderlv, 2020-09-27:
Thank you, that article is 4 years old and download links are dead.

finnzz, 2020-09-27:
Try running TuyaConvert with admin privileges, the new builds of Kali don't give you superuser rights by default which you need.

Sudo ./start_flash.sh

sanderlv, 2020-09-27:
Have done. Both on kali and ubuntu... no difference

finnzz, 2020-09-27:
Did you run with sudo? 
'Sudo ./start_flash.sh'

I don't see sudo in your screenshot

sanderlv, 2020-09-27:
Yes with sudo, but I did not include all screenshots ...

finnzz, 2020-09-27:
Sudo was the lowest hanging fruit.

Is your WiFi adapter listed as wlan0? Tuya Convert runs assuming it is.

My other thought is that your build may not have the linux drivers for the rtl8188 chipset that support AP mode.  I had to download and install an alternative driver to get this working.

I used this guide:
http://systemsnotebook.blogspot.com/2019/10/tuya-convert-on-virtualbox-with-rtl8192.html?m=1

The important takeaway there is step 5, getting and installing the rtl8192 drivers which will also work with the rtl8188 chipset.

sanderlv, 2020-09-29:
Appreciates, but it's a mess...

Trying everything everywhere... nothing helps...

```
ubuntu@ubuntu:~/rtl8192eu-linux$ ./install_wifi.sh
rmmod: ERROR: Module 8192eu is not currently loaded
rmmod: ERROR: Module rtl8xxxu is not currently loaded
Error! There are no instances of module: rtl8192eu
1.0 located in the DKMS tree.

Creating symlink /var/lib/dkms/rtl8192eu/1.0/source ->
                 /usr/src/rtl8192eu-1.0

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' all KVER=5.4.0-1019-raspi....(bad exit status: 2)
ERROR (dkms apport): binary package for rtl8192eu: 1.0 not found
Error! Bad return status for module build on kernel: 5.4.0-1019-raspi (armv7l)
Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information.
modprobe: FATAL: Module 8192eu not found in directory /lib/modules/5.4.0-1019-raspi
ubuntu@ubuntu:~/rtl8192eu-linux$

```

jjimenezlopez, 2020-10-03:
Hi,

I had this same problem today, "AP mode not supported" with the same TP-Link wifi adapter. After trying a couple of different drivers, I finally found one that allows you tu use the AP mode: https://github.com/AIRCRACK-NG/RTL8188EUS

Just follow the instructions to install it. I did it on a Ubuntu 20.4 running on VirtualBox.

I hope it helps!

jenniferlee1818, 2020-10-23:
@jjimenezlopez Thankyou!  Finally a driver that works for my TP-Link TL-WN725!

underworldgamer, 2020-12-21:
@jjimenezlopez what you have done to install it ?

This  does not work for me on Raspbery PI **2020-12-02-raspios-buster-armhf**
What kind of software does i need to install ?

echo "blacklist r8188eu" >> "/etc/modprobe.d/realtek.conf"
make && make install

BR
Mathias

josebadger, 2020-12-22:
@underworldgamer sorry, I just followed the instructions in the readme file and it worked. I did it in Ubuntu 18 on a VirtualBox machine.

jjimenezlopez, 2020-12-22:
@underworldgamer sorry, I just followed the instructions in the readme file. I did it in Ubuntu 18 on a VirtualBox machine.

Scope666, 2021-02-23:
If it helps anyone else, this driver worked for me with a RTL8811AU:

https://github.com/morrownr/8821au

codenamegary, 2021-04-03:
> Hi,
> 
> I had this same problem today, "AP mode not supported" with the same TP-Link wifi adapter. After trying a couple of different drivers, I finally found one that allows you to use the AP mode: https://github.com/AIRCRACK-NG/RTL8188EUS
> 
> Just follow the instructions to install it. I did it on a Ubuntu 20.4 running on VirtualBox.
> 
> I hope it helps!

Thank you so much! That worked perfect for me.

charliesoap4, 2021-11-21:
> Hi,
>
> I had this same problem today, "AP mode not supported" with the same TP-Link wifi adapter. After trying a couple of different drivers, I finally found one that allows you to use the AP mode: [...]

When I get to this step:
>sudo iw dev <interface> set type monitor

It says no interface.  What am I doing wrong?