ct-Open-Source/tuya-convert #771: QS-WIFI-S03

Murder3D, 2020-09-28:
Hi everyone,

I just want to report that the device **QS-WIFI-S03 Module Switch** (https://templates.blakadder.com/QS-WIFI-S03.html) appears to have been updated to the newest Tuya firmware/patch thus Tuya Convert is not working.

I bought 9 of those on May this year and I was able to flash it via Tuya Convert.

I bought one yesterday to have a spare one laying around in case of failure and I am unable to flash it.

I noticed the the "older" model emitted a fast-beeping sound when in pairing-mode while this new one has a blue light on the side instead that flashes rapidly.

I am aware that there's a pinned thread on this topic (https://github.com/ct-Open-Source/tuya-convert/issues/483) but I don't know if this issue on this specific module has already been reported so I would like to help by providing all the necessary informations which may be useful.

I am a bit of a noob on troubleshooting this because I've never had issue flashing my devices following the step-by-step guide but I can provide the informations needed if someone is willing to tell me what is needed and how to do it.

_**SETUP**_
Raspberry Pi 4
RasbianOS updated as of today
tuya-convert updated as of today
my smartphone connected to the vtrust-flash ssid

_**ISSUE**_
the procedure fails to complete the following step:
........................................
SmartConfig complete.
Autor retry in 3s. ...
Resending SmartConfig Packets
........................................

after this step normally I should see "IoT-device is online with ip 10.42.42.42" but it isn't happening.

Thank-you

kueblc, 2020-09-28:
This isn't enough information to determine the cause. Please upload your log files if you'd like help determining the issue.

Murder3D, 2020-09-29:
> This isn't enough information to determine the cause. Please upload your log files if you'd like help determining the issue.

Hi **kueblc**

thank you for taking the time to reply to this thread. 

I would like to provide all the necessary log files and informations but as I mentioned in the OP this is my first time troubleshooting such issues and I need some help on how to provide the log files needed.

Is there a link/wiki on how to do this or can you point me out in the right direction?

Many thanks

kueblc, 2020-09-29:
The `start_flash` script should tell you this.

```
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
```

Browse to the `scripts` directory and upload the files ending in `.log`

SauRoNZA, 2020-09-30:
Seeing much the same behaviour trying to flash the following.

https://templates.blakadder.com/ultralink_UL-P01W.html

First time I'm using Tuyaconvert with these devices, but I do believe it's has worked historically.

[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5303964/smarthack-wifi.log)

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5303960/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5303961/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/5303962/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/5303963/smarthack-web.log)

Murder3D, 2020-10-02:
Hey **kueblc** thanks for the suggestions.
I can configrm that after retrying this step for like 10 times

........................................
SmartConfig complete.
Autor retry in 3s. ...
Resending SmartConfig Packets
........................................

the failure message was exactly the one that you posted:

Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder

I managed to grab all the .log files that I could find

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5318876/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5318877/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/5318878/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/5318879/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5318880/smarthack-wifi.log)

Please let me know if I can help by providing other informations.

Thank you!

Murder3D, 2020-10-02:
Also, this is the inside of the device that I am trying to flash
[image]

kueblc, 2020-10-02:
@SauRoNZA @Murder3D you both are impacted by patched firmware from Tuya. It is not possible to flash these OTA at this time. See #483 and the wiki for more information.

closed 2020-10-02 by kueblc

RJSDevel, 2021-01-14:
Last known tyua swich firmware
[backup.zip](https://github.com/ct-Open-Source/tuya-convert/files/5814035/backup.zip)

pritam2016bakshi, 2023-03-08:
> Last known tyua swich firmware [backup.zip](https://github.com/ct-Open-Source/tuya-convert/files/5814035/backup.zip)

Do you have 4 channel switch firmware backups?