ct-Open-Source/tuya-convert #772: Geeni - BR30 - Prisma Drop - Failed flash

jaburges, 2020-09-29:
### Before opening a new issue, please read the following

**Are you reporting a successful flash?**
Failed Flash - multiple attempts made.

**Did you have issues getting a device to flash?**
Yes, Geeni Prisma Drop BR30 - GN-BW906-999
mqtt
```
1601399081: mosquitto version 1.5.7 starting
1601399081: Using default config.
1601399081: Opening ipv4 listen socket on port 1883.
1601399081: Opening ipv6 listen socket on port 1883.
^C1601399414: mosquitto version 1.5.7 terminating
```

psk
```
new client on port 443 from 10.42.42.24:50884
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.24:50885
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.24:50886
could not establish sslpsk socket: [SSL: WRONG_SSL_VERSION] wrong ssl version (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.24:50890
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.24:50891
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.24:50892
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.24:50893
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.24:50894
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.24:50895
could not establish sslpsk socket: [SSL: WRONG_SSL_VERSION] wrong ssl version (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.24:50896
could not establish sslpsk socket: [SSL: WRONG_SSL_VERSION] wrong ssl version (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.24:50898
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.24:57975
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.24:57417
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.24:61843
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.24:62366
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.24:60266
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.24:62194
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
^CTraceback (most recent call last):
[... 122 lines cut ...]
hostapd v2.8-devel
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2019, Jouni Malinen <j@w1.fi> and contributors
Python 3.7.3
===========
Attempting to stop wpa_supplicant
Configuring AP interface...
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:95:43:24 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
wlan0: AP-STA-CONNECTED 66:19:90:1b:d4:8a
wlan0: interface state ENABLED->DISABLED
wlan0: AP-STA-DISCONNECTED 66:19:90:1b:d4:8a
wlan0: AP-DISABLED 
wlan0: CTRL-EVENT-TERMINATING 
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...
```

kueblc, 2020-09-29:
Does the device exit pairing mode?

It looks like your iPhone is in a reconnect cycle. Apple makes the assumption that you only want to stay connected to a network with internet access, which by design vtrust-flash does not. See if you can force it to stay connected or try another device.

I see another device, `66:19:90:1b:d4:8a`, but an OUI search returns no results. Is this the device?

jaburges, 2020-09-29:
i turned of iOS private devices to avoid randomizing the MAC address.
Now you mention it I did use an android device last time! 
brb

jaburges, 2020-09-29:
so tried on android (and restarted the pairing mode) and we have a successful update :) Thanks for the info! 
Closing and i'll update the title for others

jaburges, 2020-10-04:
balls - ordered 2 more from Amazon and these may have been patched
The bulb exits pairing mode but start_flash.sh simply repeats the "Resending SmartConfig Packets" 

wifi:
```
Configuring AP interface...
RTNETLINK answers: File exists
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:95:43:24 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
wlan0: AP-STA-CONNECTED 7c:46:85:d7:96:cb
wlan0: AP-STA-CONNECTED 68:57:2d:d7:b1:0d
wlan0: AP-STA-CONNECTED 68:57:2d:d7:ad:9d
wlan0: AP-STA-DISCONNECTED 68:57:2d:d7:b1:0d
wlan0: interface state ENABLED->DISABLED
wlan0: AP-STA-DISCONNECTED 68:57:2d:d7:ad:9d
wlan0: AP-STA-DISCONNECTED 7c:46:85:d7:96:cb
wlan0: AP-DISABLED
wlan0: CTRL-EVENT-TERMINATING
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...
```
MAC `7c:46:85:d7:96:cb` is my android device

psk
```
new client on port 443 from 10.42.42.34:46300
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.34:59841
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.34:38526
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.34:47539
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.34:45264
could not establish sslpsk socket: [SSL: WRONG_SSL_VERSION] wrong ssl version (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.34:45357
could not establish sslpsk socket: [SSL: WRONG_SSL_VERSION] wrong ssl version (_ssl.c:1056)
don't panic this is probably just your phone!
```

any thoughts?