ct-Open-Source/tuya-convert #773: Teckin SS31 did not appear with the intermediate firmware

qcasey, 2020-09-29:
I think this has been noted before, but after taking down the AP and bringing it back up again, the device lights up green as it automatically reconnects. Still no luck flashing though.

Logs are attached, I will try to open this weekend and get firmware dump

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5301693/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5301694/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/5301695/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/5301697/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5301700/smarthack-wifi.log)

dshanske, 2020-10-01:
Have this problem as well

kueblc, 2020-10-01:
Unfortunately you've been impacted by patched firmware, OTA flashing is currently not possible.

See #483 and the wiki for more information.

closed 2020-10-01 by kueblc