ct-Open-Source/tuya-convert #774: Syska 12W smart bulb - SmartConfig retries and then comes with "Device did not appear with intermediate firmware"

raghur, 2020-09-30:
**Did you have issues getting a device to flash?**
Device - Syska 12W smart bulb 
https://www.amazon.in/gp/product/B08BKWMBBJ/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

** What happens

1. PHone connects to `vtrust-flash`
1. LIght blinks rapidly post boot. Once I start `start_flash`, it goes solid, reports sending SmartConfig packets and then comes up wtih 
```
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
................................................................................................................
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
```
[tuyalogs.zip](https://github.com/ct-Open-Source/tuya-convert/files/5304611/tuyalogs.zip)

So one question I have is that the wifi log I see the mac id of the bulb, but I don't see any address being assigned to it. Also, other than the SmartConfig messages after which the bulb light becomes solid, all the logs seems pretty quiet... Is it even doing anything or is something possibly blocking it.

kueblc, 2020-09-30:
Unfortunately not ESP8266 based and thus not supported by tuya-convert or third party ESP firmware like Tasmota.

closed 2020-09-30 by kueblc

raghur, 2020-09-30:
Aah - I should have checked the mac address... Sorry- my bad.