ct-Open-Source/tuya-convert #775: Rapsberry Pi4B - not working

sanderlv, 2020-09-30:
So I had no luck with a pi2 and dlink wifi, ordered a Pi4 with wifi built in. No luck yet.

Installed all to the letter but getting an error:

```
kali@kali:~/tuya-convert$ sudo ./start_flash.sh
tuya-convert v2.4.4
Checking for network interface wlan0... Found.
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen....
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
Traceback (most recent call last):
  File "./smartconfig/main.py", line 26, in <module>
    smartconfig( passwd, ssid, region, token, secret )
  File "/home/kali/tuya-convert/scripts/smartconfig/smartconfig.py", line 46, in smartconfig
    sock = SmartConfigSocket()
  File "/home/kali/tuya-convert/scripts/smartconfig/smartconfig.py", line 29, in __init__
    self._socket.bind((address, 0))
OSError: [Errno 99] Cannot assign requested address
........................................................................................................................
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N]
```
Whats wrong?

kueblc, 2020-09-30:
Answered on #745

sanderlv, 2020-09-30:
Going further on this issue :-).

What files do I need to check?

This is my device btw.

https://github.com/xoseperez/espurna/wiki/Hardware-NEO-COOLCAM-NAS-WR01W

kueblc, 2020-09-30:
Please try to help yourself before seeking assistance from others. All of the information you are requesting is readily available. I do not have time to walk you through it, read the instructions you are presented.

sanderlv, 2020-09-30:
> 
> 
> Please try to help yourself before seeking assistance from others. All of the information you are requesting is readily available. I do not have time to walk you through it, read the instructions you are presented.

I am reading for days! tried 3 different hardware setups, seeking, reading, googling. Please spare me these type of messages. I really appreciate all the help I can get. As this is not very easy for me. I would not even know where to start reading...

Here is nothing about my error:
https://github.com/ct-Open-Source/tuya-convert

Here is no information that I can find due to the fact I do not know how to look for it (what error message)
https://github.com/ct-Open-Source/tuya-convert/issues

Here is no information about my issue...
https://github.com/xoseperez/espurna/wiki/Hardware-NEO-COOLCAM-NAS-WR01W

See...

What I hope is that I can flash my device via this cool tool (invested 5 times more in the equipment than the actual neo coolcam wifi switched costed, just because I like the challlenge... well untill it works :-)

sanderlv, 2020-10-01:
Updated with latest progress here. Please guide me on how to get the logfiles from the pi to github.

https://github.com/ct-Open-Source/tuya-convert/issues/666#issuecomment-702066538