ct-Open-Source/tuya-convert #776: New WF-DS01 will not update with tuya-convert

gwietze, 2020-10-01:
### Before opening a new issue, please read the following

**Are you reporting a successful flash?**
Not with tuya-convert, with serial-connection it will flash
In the Zip file, I have a backup of the tuya firmware maybe you can look for a new PSK file?
The backup is done with tasmotizer.

[backup_20201001_190724.zip](https://github.com/ct-Open-Source/tuya-convert/files/5314496/backup_20201001_190724.zip)

kueblc, 2020-10-01:
Hi @gwietze, thanks for the report.

The binary you uploaded is empty (entirely `0xFF`). Can you check that you uploaded the correct backup?

gwietze, 2020-10-03:
I have another one for you, I think this one has firmware.
[backup_20201001_180232.zip](https://github.com/ct-Open-Source/tuya-convert/files/5321917/backup_20201001_180232.zip)

jschwalbe, 2020-10-10:
I've been trying for days to get this to flash via serial, but am having zero success. Can either of you guys help? I've tried cutting the 3V to the chip (thus hopefully disabling the MCU), cutting RX/TX from the ESP (again to disable MCU from interfering w/ flashing) but I am having zero luck. Have you been able to flash this?

jschwalbe, 2020-10-13:
@kueblc this is slightly off-topic (i'm happy to move to another discussion medium) -- but is there any place for like-minded people who might be interested in lending a hand to combat the new tuya to hang out? discord or otherwise? i've uploaded pre/post firmwares + pcap to the wiki, and can upload more if interested, but i haven't heard otherwise.

kueblc, 2020-10-13:
@jschwalbe I'm glad you asked, I had been meaning to reach out. I was your upload but unfortunately I couldn't find any pskKey written to flash. I think you're right, it would be helpful to have more real time chat.

I'm not sure what we should use, perhaps Telegram or Signal? I do not have Discord set up but I can look into it.

digiblur, 2020-10-13:
Definitely doable if you want some sort of channel with a few folks let me know. We already have a public Tuya convert channel for regular stuff.

gwietze, 2020-10-13:
WF-DS01 can be flashed but you have to take it slowly , 9600 baud I had succes with flashing 
And look at the picture on this page https://github.com/arendst/Tasmota/issues/4543

jschwalbe, 2020-10-16:
@kueblc https://discord.gg/cPAxGqm if you're interested

gwietze, 2020-11-21:
I have now a workaround to program the WF-DS01 on serial. It's not nice but it worked for me.

I have four of them and the first two was not a problem to program but the other two was hell. Losing connection and can't detect the chip. Half way of programming and then errors. I have done a lot of experiments  with this problem. I have changed the esp but that doesn't make any differents. I pulled out the board so I can put on a external powersupply but no resold. But on the power of the serial to usb converter sometimes it worked and some times not. (unstable)
Then I connect it back on the external powersupply and then on 3.3 volts and it wont program. Then I put the voltage down to 2.9 volt and there it was... 
Conclusion: When you pull the board out and put 2.8V on pin VCC on the esp then it will program on serial.
and with a higher speed. I have tested it on 115200.
I think,  the other MCU STM8S003F3 wont start with low voltage, so the serial port of the esp is free to go.

pritam2016bakshi, 2023-03-09:
> I have another one for you, I think this one has firmware. [backup_20201001_180232.zip](https://github.com/ct-Open-Source/tuya-convert/files/5321917/backup_20201001_180232.zip)
Both files are don't have any firmware.