ct-Open-Source/tuya-convert #778: Failed flash of BN-LINK BNC-60/U133TJ

OriginalOC, 2020-10-01:
Picked up 4 of these this afternoon & failed on the flash of them. Lots of "could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC]" entries in 'smarthack-psk.log'. Attached are the logs.

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5314689/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5314690/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/5314691/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/5314693/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5314694/smarthack-wifi.log)

kueblc, 2020-10-01:
Duplicate of #483. Unfortunately these are not currently flashable OTA due to a patch by Tuya. See wiki for more information.

closed 2020-10-01 by kueblc

RGSSoftware, 2020-10-06:
@OriginalOC as of 10/03/2020 it seems the [3 pack](https://www.amazon.com/dp/B07HPG58FP/ref=twister_B07VDGM6QR?_encoding=UTF8&th=1) on Amazon still has the old un-patched version. I've flashed the 3 pack on 10/05/2020 with tuya-convert successfully. I've purchased the [4 pack](https://www.amazon.com/dp/B07CX5KLXN/ref=twister_B07VDGM6QR?_encoding=UTF8&th=1) on 09/16/2020 and it successfully flashed with tuya-convert. I've purchased another 4 pack on 10/02/2020 but it had the "could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC]" entries in 'smarthack-psk.log'. So anybody reading this there's maybe a small window to get the [3 pack](https://www.amazon.com/dp/B07HPG58FP/ref=twister_B07VDGM6QR?_encoding=UTF8&th=1) on Amazon before they fail with tuya-convert.

OriginalOC, 2020-10-06:
Thanks for mentioning that! I've just placed an order & should receive it
tomorrow.

zervin, 2020-10-06:
Sonofa...just received some today..didn't check her for issues prior...I wouldn't bother ordering anymore of these folks. Anyone lay eyes on "still working" 15A outlets?

OriginalOC, 2020-10-09:
I received my 3-pack yesterday & just finished flashing them over morning coffee. Thanks @RGSSoftware for the suggestion to try the 3-pack version over the 4-pack, which I and others have had problems flashing. In the last 2 weeks, these have been my successes:

https://smile.amazon.com/gp/product/B07HPG58FP/
https://smile.amazon.com/gp/product/B07JW66S37/
https://smile.amazon.com/gp/product/B07PF41CLZ/

NeoMatrixJR, 2020-10-17:
Dammit...bought some on prime day 😢 
So...SOL...send 'em back?
Is there a chat/development thread on this?  Starting to get worried I'm going to have a hard time finding IoT devices I can put clean HA friendly open source firmware on.....

AlecLetourneau, 2020-10-19:
Just went to order another 3 pack of these and found that the 3 pack is no longer listed.
There's a good chance that future 3 packs will be patched now. 😕

netstx, 2020-11-24:
Just to add here, I purchased 2x 4-packs on November 12, 2020 and have not been able to flash them. I get the following error on smarthack-psk.log:

```
new client on port 443 from 10.42.42.32:39354
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
```

RGSSoftware, 2020-11-24:
I think these plugs are a '**no go**' moving forward. The 3 pack is no longer listed on amazon and hasn't been since 10/18/2020. I suspect the 3 pack didn't sell well and will not be restocked, only the 2 and 4 pack will (with **patched firmware**).

netstx, 2020-11-24:
@RGSSoftware yeah, I think so too... it's a bummer. I did find this in my logs, I wonder if it's expected if the process was working? I wonder if I am doing something wrong possibly. I hold the power button down until it starts blinking, but nothing really happens.

```
Listening on 10.42.42.1:80
^[32m[I 201124 11:18:36 web:2243]^[m^ 302 GET /mobile/status.php (10.42.42.32) 3.35ms
^[32m[I 201124 11:18:36 web:2243]^[m^ 200 GET / (10.42.42.32) 5.10ms
^[32m[I 201124 11:22:11 web:2243]^[m^ 302 GET /gen_204 (10.42.42.32) 1.74ms
^[32m[I 201124 11:22:11 web:2243]^[m^ 302 GET /generate_204 (10.42.42.32) 1.86ms
^[32m[I 201124 11:22:23 web:2243]^[m^ 302 GET /gen_204 (10.42.42.32) 2.88ms
^[32m[I 201124 11:22:23 web:2243]^[m^ 200 GET / (10.42.42.32) 2.69ms
^[32m[I 201124 11:22:23 web:2243]^[m^ 302 GET /favicon.ico (10.42.42.32) 2.71ms
^[32m[I 201124 11:22:23 web:2243]^[m^ 304 GET / (10.42.42.32) 2.66ms
^[32m[I 201124 11:22:24 web:2243]^[m^ 302 GET /gen_204 (10.42.42.32) 5.47ms
^[32m[I 201124 11:22:24 web:2243]^[m^ 302 GET /gen_204 (10.42.42.32) 5.48ms
^[32m[I 201124 11:23:07 web:2243]^[m^ 302 GET /gen_204 (10.42.42.32) 3.11ms
^[32m[I 201124 11:23:07 web:2243]^[m^ 200 GET / (10.42.42.32) 2.96ms
^[32m[I 201124 11:23:08 web:2243]^[m^ 302 GET /favicon.ico (10.42.42.32) 2.58ms
^[32m[I 201124 11:23:08 web:2243]^[m^ 304 GET / (10.42.42.32) 2.61ms
^[32m[I 201124 11:23:09 web:2243]^[m^ 302 GET /gen_204 (10.42.42.32) 3.28ms
^[32m[I 201124 11:23:09 web:2243]^[m^ 302 GET /gen_204 (10.42.42.32) 2.84ms
```

kueblc, 2020-12-02:
@netstx

> ```
> new client on port 443 from 10.42.42.32:39354
> could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
> ```

This is probably not your IoT device, but rather the device you connected to begin smart config, smart phone etc.

The second log you posted is activity from an Android phone with the same IP address. It does not tell us anything about your IoT device. See if the device connected at all by checking the wifi log.

luke-jr, 2021-01-25:
BNC-60 is actually NOT a case of #483

I pulled one of these apart and found a BK7231 (ARM/Thumb) instead of an ESP (Xtensa) chip.

So more like #692