ct-Open-Source/tuya-convert #780: TYWE3S fail

dbrokenbone, 2020-10-04:
i found out that my tuya switch 3gang(wifi) using TYWE3S, and i having problem flashing it with tasmota, can it flash with tasmota?

oldschoolharry, 2020-10-05:
I have the same issue... getting the 'Resending Smartconfig packets' looping but then nothing.

finnzz, 2020-10-05:
If you upload your log files, we can better determine what the problem may be.

drudgebg, 2020-10-06:
Same with my new Zemismart 3gang Switch with TYWE3S
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:852)

kueblc, 2020-10-06:
Please fill out the requested information when opening a new issue. We can't help you if you don't give us any information.

closed 2020-10-06 by kueblc

dbrokenbone, 2020-10-07:
> I have the same issue... getting the 'Resending Smartconfig packets' looping but then nothing.

ok solved that by flashing it with usbttl cable and gpio0 to ground, and flash it with 'tasmota.bin'

dbrokenbone, 2020-10-07:
> Same with my new Zemismart 3gang Switch with TYWE3S
> could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:852)

flash it with usbttl is way easier tho...im on Bardi's hardware

[image]

drudgebg, 2020-10-07:
Thanks, I already do it too. Now everything is fine.

> > Same with my new Zemismart 3gang Switch with TYWE3S
> > could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:852)
> 
> flash it with usbttl is way easier tho...im on Bardi's hardware
> 
> [image]

dbrokenbone, 2020-10-08:
> Thanks, I already do it too. Now everything is fine.

awesome...good to knw...