ct-Open-Source/tuya-convert #782: tuya-convert hangs after start_flash.sh at "starting web server in a screen"

PuckeyP, 2020-10-07:
Hi All,
I have a problem using tuya-convert.
I have installed VMWarePlayer on an up to date windows machine, installed Ubuntu Desktop, and installed tuya-convert conform instructions. All goes well, even with exactly the expected output like in the install examples, but the installation/execution stops at  "starting web server in a screen".
Any ideas how to solve this?
PPee

PuckeyP, 2020-10-07:
mmmm... sorry. should have been more patient

PuckeyP, 2020-10-07:
Still having problems with this. Takes such a long time for "starting webserver". This doesnot look normal is it?

kueblc, 2020-10-07:
It should not take more than a second or two to start the web server. Check your smarthack-web.log file for issues.

PuckeyP, 2020-10-07:
Thanks Colin,
Must be something wrong with VMWarePlayer 
Switched to a Pi (works fine),and discovered that my Tuya power switches are based on a tot supported chipset 'winner'. Thx for the swift response
PPee

kueblc, 2020-10-08:
No problem, sorry to hear your devices are not flashable.

closed 2020-10-08 by kueblc