ct-Open-Source/tuya-convert #784: 3A downlight - never joins wifi, but responds to smartconfig packets?

takigama, 2020-10-08:
So this next device is a 3A downlight, physically almost identical to the genio downlight, but never appears to actually join the wifi at all and yet as soon as smartconfig packets are sent, it drops out of EZ config mode (i.e. if i wait at the "3. Press ENTER to continue" prompt for any random length of time, the device stops flashing shortly after hitting enter)...

In the logs all I can see is this (my phone joining the wifi):
wlp59s0: AP-STA-CONNECTED de:0a:09:18:a0:15

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5345916/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5345917/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/5345918/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/5345919/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5345920/smarthack-wifi.log)

ctaetcsh, 2020-10-31:
Having the same issue

toasterlint, 2020-11-03:
I’m also seeing similar behavior with Lumiman lightbulbs. Only device that shows that joins the WiFi network is my phone, yet the lightbulb is somehow receiving the smartconfig packets to drop it out of EZ config mode.

fraintt, 2020-12-23:
I have the same issue with a rgbcct 50w flood light bought from aliexpress. As soon as I press Enter at stage 3. the light stops flashing but no second device connected in the smarthack-wifi.log