ct-Open-Source/tuya-convert #786: New Device - Device did not appear with the intermediate firmware

tobit6795, 2020-10-10:
Hi,

i just found a new tuya device that i am trying to flash but failed:

https://www.amazon.de/gp/product/B0893T8JPM/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1

Logs:
[
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5358862/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5358863/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/5358864/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/5358865/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5358866/smarthack-wifi.log)

](url)

I've seen that some informations were discovered (hopefully useful?!)

10.42.42.32 {"ip":"10.42.42.32","gwId":"72778583fcf5c48371de","active":2,"ability":0,"mode":0,"encrypt":true,"productKey":"lyq7o3cpui0mgfky","version":"3.3"}
^C

Let me know if i can provide any other information

kueblc, 2020-10-10:
Looks like it is nearly there. The first attempt was stopped just before it could download the firmware, the second attempt looks like a hang while downloading the firmware. Third time's the charm?

tobit6795, 2020-10-10:
LOL. i reduced the distance for the second time, it finally worked! 
[20201010_140555.zip](https://github.com/ct-Open-Source/tuya-convert/files/5359172/20201010_140555.zip)

Now i need to figure out how to configure the template...

kueblc, 2020-10-10:
Great, glad it worked out

closed 2020-10-10 by kueblc