ct-Open-Source/tuya-convert #787: failed flash attempt on Feit OM60/927CA/AG

rca, 2020-10-10:
i ran into the [tasmota compatibility](https://templates.blakadder.com/bulb.html) page that says the `OM60/927CA/AG` is compatible.  i picked up a three pack at my local hardware store (Lowe's) but when trying tuya-convert i get a message that the device doesn't use an ESP8266.

i'm wondering if new iterations of the device are truly no longer ESP devices or if the previously hackable entrypoint has been patched up and this tool needs adjusting.

the web logs are below; please let me know if I can provide any additional information.

thanks!

### Before opening a new issue, please read the following

**Are you reporting a successful flash?**

no

**Do you have a question about configuring your newly flashed device?**

no

**Did you have issues getting a device to flash?**

yes, Feit `OM60/927CA/AG/3`; looks just like `OM60/927CA/AG` but sold in a 3-pack

not sure what firmware version you mean.  i cloned the tuya-convert repo:

```
commit 94acc3bb020361266ceb74f082e1d25c92a60345
tag: v2.4.4 
```

logs:

```
POST /d.json?a=tuya.device.config.get&et=1&other={"token":"00000000","region":"US"}&t=10&uuid=4eab1083213a3300&v=4.1&sign=21337e669c59522d8f45131f76255182
Host: a2.tuyaus.com
User-Agent: TUYA_IOT_SDK
Connection: keep-alive
Content-Type: application/x-www-form-urlencoded; charset=UTF-8
Content-Length: 37

payload D811108D2B060B42DBD1FB797AB8FC76
WARNING: it appears this device does not use an ESP82xx and therefore cannot install ESP based firmware
Answer tuya.device.dynamic.config.get
reply {"result":"fn8oBaEAaJe3I+enO7OkYdd0LkD9WHnY6k8vmvctrJYKyHK9baFkODoII//qn92v9F2VBmqNqTqJcWuiX4YubAQ3yqMWR+9ej6R/WWEcpeO7mevrGF2WUpAllBDW/Q5l","t":1602300761,"sign":"2e1b4dd8cb57e67e"}
ESC[32m[I 201009 23:32:41 web:2246]ESC[m^O 200 POST /d.json?a=tuya.device.config.get&et=1&other={"token":"00000000","region":"US"}&t=10&uuid=4eab1083213a3300&v=4.1&sign=21337e669c59522d8f45131f76255182 (10.42.42.1) 4.50ms

POST /d.json?a=tuya.device.config.get&et=1&other={"token":"00000000","region":"US"}&t=11&uuid=4eab1083213a3300&v=4.1&sign=42d9912db03519a321d3f7bc141446ea
Host: a2.tuyaus.com
User-Agent: TUYA_IOT_SDK
Connection: keep-alive
Content-Type: application/x-www-form-urlencoded; charset=UTF-8
Content-Length: 37

payload 0EF96B85A64885366D17D83FCED47402
WARNING: it appears this device does not use an ESP82xx and therefore cannot install ESP based firmware
Answer tuya.device.dynamic.config.get
reply {"result":"fn8oBaEAaJe3I+enO7OkYdd0LkD9WHnY6k8vmvctrJaKkDVs4f3X6chWit+GAlJJ9F2VBmqNqTqJcWuiX4YubMoDXL//1/PMdPxuPmnGvyO7mevrGF2WUpAllBDW/Q5l","t":1602300762,"sign":"76a854f16b597c98"}
ESC[32m[I 201009 23:32:42 web:2246]ESC[m^O 200 POST /d.json?a=tuya.device.config.get&et=1&other={"token":"00000000","region":"US"}&t=11&uuid=4eab1083213a3300&v=4.1&sign=42d9912db03519a321d3f7bc141446ea (10.42.42.1) 3.41ms
```

kueblc, 2020-10-10:
Yes, sorry to say, the device detection is accurate. Your bulbs do not use the ESP82xx. Manufacturers have been switching over existing models to new chipsets for some time now. The template repository is not a reliable indicator of support at this time.

closed 2020-10-10 by kueblc

rca, 2020-10-10:
That's unfortunate, thank you for the update.  I bought a single pack just in case they were different and it also fails with the same log messages.  The box is stamped with `(c) 2019 Feit Electric, Rev 092419H-228`.  Presumably any bulb created after September 24th 2019 will no longer work.  :disappointed: