ct-Open-Source/tuya-convert #792: Arlec Grid Connect Smart Plug Fails Backup

r0kk3rz, 2020-10-12:
Arlec Grid Connect Smart Plug times out during the backup process, and then becomes unresponsive to further requests.

after disconnecting and reconnecting device and commenting out backup process, tasmota flashed successfully and is working ok

r0kk3rz, 2020-10-12:
I added a note to the Troubleshooting Wiki page, describing what I saw and what I did to remedy

In general it might be nice if the intermediate firmware used the onboard LED to show some kind of state, like On,Connected to WIFI, and Failure modes. Otherwise the thing just looks completely bricked until you go searching for extra Wifi SSIDs out there

kueblc, 2020-10-12:
> after disconnecting and reconnecting device and commenting out backup process, tasmota flashed successfully and is working ok

I may have an idea of why this is happening, but I will need a backup from such a device to test this theory.

> I added a note to the Troubleshooting Wiki page, describing what I saw and what I did to remedy

Awesome, thank you

> In general it might be nice if the intermediate firmware used the onboard LED to show some kind of state, like On,Connected to WIFI, and Failure modes. Otherwise the thing just looks completely bricked until you go searching for extra Wifi SSIDs out there

I'd love to be able to do this, but unfortunately we have no idea what kind of hardware the intermediate firmware is running on. This is designed to work on a huge variety of devices. The LED might be on a different pin or there might be none at all. Flipping pins on unknown hardware could be disastrous.

r0kk3rz, 2020-10-12:
> 
> I may have an idea of why this is happening, but I will need a backup from such a device to test this theory.
> 

I have several partial backups if that helps, no full backup was obtained

kueblc, 2020-10-13:
Do you have any remaining troublesome devices still running the stock firmware?

r0kk3rz, 2020-10-28:
I might buy another one at some point, but in the mean time i've opened the case and it has the TYWE2S tuya board in it if that helps at all.

https://developer.tuya.com/en/docs/iot/device-development/module/wifi-module/wifie2smodule?id=K9605u79tgxug