ct-Open-Source/tuya-convert #793: How to connect USB WiFi adapter when using Docker?

QuantiumDev, 2020-10-12:
In the Using Docker section of the readme, it would be really helpful if there was a link to a page that explains how to identify an attached USB WiFi adapter and pass it through to the container in the docker-compose.yml file. I found the section that I need to add to docker-compose.yml
`    devices:
      - "/dev/ttyUSB0:/dev/ttyUSB0"`
but I can't figure out how to identify the USB NIC in order to properly configure those tty values.

kueblc, 2020-10-13:
To be honest I am not sure. The docker config was contributed by a community member and is not really the "official" method, and since I do not use Docker I can't provide support.

If you do find out, please let us know. I would be happy to accept a PR with updated instructions for this.

pdugas, 2020-10-14:
`ip link` or `ifconfig` should show you the interfaces on your host machine. You should see the same within the container AFAIK. Given the `privilege` setting in the Compose file, all of those interfaces should be usable in the container. It will not be the `ttyUSB0` device you referenced - that's a serial UART, not a network interface. Use `docker-compose exec tuya /bin/sh` to get a shell in the running container to poke around.

Cyber1000, 2021-01-30:
@QuantiumDev The information from @pdugas is right, you don't have to pass a device to the container, just adjust the WLAN in your docker-compose.yml to the right wlan-interface (wlan0 is the default here, but your network may differ)
ip link and ifconfig is a good start, some other commands you may find here: https://www.cyberciti.biz/faq/linux-list-network-cards-command/

phaseOne, 2021-06-17:
You would use a [`device` directive](https://docs.docker.com/compose/compose-file/compose-file-v3/#devices) in your `docker-compose.yml` file. It is equivalent to the [`--device` option in the Docker CLI](https://docs.docker.com/engine/reference/commandline/run/#add-host-device-to-container---device). Note that this is currently only possible on Linux hosts, as it's [not supported in Docker Desktop for macOS or Windows](https://docs.docker.com/desktop/faqs/#can-i-pass-through-a-usb-device-to-a-container) (see also https://github.com/docker/for-mac/issues/900), though there is [a potential workaround using VirtualBox](https://christopherjmcclellan.wordpress.com/2019/04/21/using-usb-with-docker-for-mac/#tldr) that I haven't tested.

Attaching a USB NIC via Docker seems to be more trouble than it's worth, and would be better to boot your main system into a Linux image on USB or to use a Raspberry Pi.

Milor123, 2025-03-30:
Have you guys made any progress on this in 2025? Is there any way we can do it now?