ct-Open-Source/tuya-convert #796: Alternative Dockerfile

pdugas, 2020-10-14:
I'm a little braindead today and didn't notice the project includes a Dockerfile before I'd gone ahead a rolled my own. It's simpler and builds the image *much* faster than the Ubuntu image you use currently so I thought I'd share in case it's of interest.  Feel free to ignore this if not. There was only one hack to the existing project code needed to get it to work; busybox's `ls` doesn't support the `-m` option. 

```Dockerfile
FROM alpine:latest
RUN apk add --update bash git iw dnsmasq hostapd screen curl py3-pip py3-wheel python3-dev mosquitto haveged net-tools openssl openssl-dev gcc musl-dev linux-headers sudo
RUN python3 -m pip install --upgrade paho-mqtt tornado git+https://github.com/drbild/sslpsk.git pycryptodomex
RUN git clone --depth 1 https://github.com/ct-Open-Source/tuya-convert /usr/local/tuya-convert
RUN sed -i 's|ls -m|ls|' /usr/local/tuya-convert/scripts/setup_checks.sh
WORKDIR "/usr/local/tuya-convert"
```

A companion Makefile too...

```Makefile
NAME="tuya-convert"
EXEC ?= /bin/sh
run:
	@if [ "$(shell docker ps -q -f "name=$(NAME)")" = "" ] ; then \
		docker build --tag $(NAME) .; \
		docker run --rm -it --network=host --name=$(NAME) --privileged $(NAME) $(EXEC); \
	else \
		docker exec -it $(NAME) $(EXEC); \
	fi
clean:
	@[ "$(shell docker ps -q -f "name=$(NAME)")" = "" ] || docker stop $(NAME)
	@docker rmi $(NAME)
.PHONY: run clean
```

Hurricos, 2020-10-28:
Fantastic. This works great!

I hate that my Python userland is too new to do this natively. I love that I can just download a new userland and play in that :)

pdugas, 2020-10-28:
Head's up... The `check_port` logic in `setup-checks.sh` won't work when running in a container like this. You will need to ensure the corresponding local services are disabled on the host machine yourself.

Hurricos, 2020-10-28:
> Head's up... The `check_port` logic in `setup-checks.sh` won't work when running in a container like this. You will need to ensure the corresponding local services are disabled on the host machine yourself.

Good mention. Folks who try this the first time after running tuya outside of a container won't have noticed.

tablatronix, 2020-11-26:
branch or PR for this?

Cyber1000, 2021-01-29:
Hi @pdugas, 
I originally added the docker-image in this project, thanks for your work!
The main reason I choose an ubuntu-based-docker-image was that  I could use install_prereq.sh, without always have to separately adjust pacakges.
Perhaps we can take https://github.com/ct-Open-Source/tuya-convert/pull/890 to extend install_prereq.sh to also detect alpine_images ...
I'll have a look into this in the next few days ...