ct-Open-Source/tuya-convert #797: Tuya 1 Gang Switch (Control with MQTT broker configure with Tasmota)

M-Ahmad-Naeem, 2020-10-15:
Hi!

I am playing with Tuya 1 Gang Switch for a while. I have successfully flashed a Tasmota andcontorl the switch with WebUI.
Now I am trying to do the same with MQTT broker.
I need some guidance in this regard.

Thanks for support in advance.

kueblc, 2020-10-15:
Please see the Tasmota community for Tasmota configuration support

closed 2020-10-15 by kueblc

M-Ahmad-Naeem, 2020-10-16:
> Please see the Tasmota community for Tasmota configuration support

I am afraid, I have looked at the support and was unable to find a definite answer to my problem.
So turned to Github community. 
Little help is highly appreciated
Thank your for your response.

kueblc, 2020-10-16:
This is tuya-convert, we do not produce Tasmota. You can find the Tasmota repository on github as well.

M-Ahmad-Naeem, 2020-10-20:
Hi!

I have successfully controlled the 1 GANG switch with MQTT broker.
I used MQTT.fix and HiveMQ broker for controlling the switch.

Regards