ct-Open-Source/tuya-convert #800: cant flash lsc connect gu10

tobit6795, 2020-10-17:
hi there again,

i was trying to flash some new buld from lsc gu10 which are listes as working devices. i already got them flashed half a year ago, so i buyed them again.

at the moment im not able to flash them

[logs.zip](https://github.com/ct-Open-Source/tuya-convert/files/5396856/logs.zip)

kn100, 2020-10-21:
I'm not the maintainer or related to this repo but you're probably affected by https://github.com/ct-Open-Source/tuya-convert/wiki/Collaboration-document-for-PSK-Identity-02 - of which there is no solution yet except for hardware level flashing.

GRaptor, 2021-02-24:
Okay, similar story with me. Bought about 22 of the LSC GU10 RGBW type in three batches. They show these part numbers/versions:

1. 970702 v1.0 Type AL K19c (bought End 2019)
2. 970702 v2.0 Type AL K19c (bought mid 2020)
3. 970702.1 v1.0 Type AL G20b (bought end 2020)

Out of batches 1 and 2, 16 worked to flash, with four I had different issues (eg worked only randomly or showed wrong colors)

Batch 3 (2 bulps) shows what's described in this issue. My screen output and log files are:
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/6038636/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/6038637/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/6038638/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/6038639/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/6038640/smarthack-wifi.log)
[Screen dump.txt](https://github.com/ct-Open-Source/tuya-convert/files/6038665/Screen.dump.txt)

Taking the type name G20b of batch 3, it sounds like what's described in #855 .
However, my smarthack-psk.log shows different information and the screen output says it may be an ESP82xxx
I know, this may missleading information. But it point's to this issue here.

Any ideas?

Fyi, as I have no use for the bulps w/o flashing and it's too late to return them, I'm willing to sacrifice one if this supports problem solving.

GRaptor, 2021-03-27:
So, today I opened one of the LSC GU10  970702.1 v1.0 versions.
It still does have an ESP8285.
However, what's next? Does it mean new firmware?