ct-Open-Source/tuya-convert #802: Ultrabrite branded Tuya smart plug

kn100, 2020-10-18:
Hi, I recently bought an Ultrabrite branded Tuya smart plug - which definitely has some ESP thing inside of it (see board shots). I can't seem to get tuya-convert to flash. I've tried both on my desktop and a Raspberry pi 3b+ so I don't think it's wifi related - it seems as if the plug doesn't recognise the smartconfig bytes being sent to it - since the next step in the code is to ping the device continuously until it replies. 

Board pics: https://imgur.com/a/fpGSPLq
First set of logs: https://pastebin.com/BafY6ki8
Main module: 1.1.0
MCU module: 1.1.0

Here is an attempt I did to get clean logs to share.
1. The plug is connected and works via the Ultrabrite app.
2. I run ./start-flash.sh with sudo
3. Dnsmasq is terminated
4. Mosquitto is terminated
5. I connect my Huawei p30 to the vtrust-flash network
6. I hold the power button on my plug for 5 seconds, which causes the green LED to blink rapidly
7. I press enter on the warning
8. The plug goes solid green after the first set of SmartConfig bytes gets sent, but the tool continually retries - I suppose because the plug isn't connecting to the AP.
9. Device did not appear with the intermediate firmwareurl
10. I export the logs (see the first set of logs!)
11. I say no to the flashing another device
12. I start ./start-flash.sh with sudo again
13. systemd-resolve is terminated
14. I do NOT put the plug into config mode nor do I connect my phone to the AP
15. I press enter to continue
16. Once again, connection fails.
17. I export the logs - if you want these let me know!
18. I say no to the reflash question again.

Any idea what might be wrong? I really don't want to try to solder to this board - it's super awkward in there!

kn100, 2020-10-18:
I should say if you need a firmware dump or something I am happy to solder to the serial port internally - but it'll take me a little while.

kn100, 2020-10-20:
Ah, It's probably https://github.com/ct-Open-Source/tuya-convert/wiki/Collaboration-document-for-PSK-Identity-02. That sucks. If I can help at all please let me know. Happy to destroy a plug or two in the name of fighting off this IOT protectionist nonsense - just let me know how I can be of use!

Pr0jectN1ne, 2020-10-21:
im having the same issue with a bakibo tp24y. hope someone can help.

eccen, 2020-10-29:
I'm getting the same behavior with a BN-LINK BNC-60/U133TJ, which is listed as "working" on the list.

kn100, 2020-10-29:
To add to this discussion, after doing some rather destructive modification to these plugs, they flash just fine with Tasmotizer and a serial programmer, so it is almost certainly this issue. I have a firmware dump but no captured network traffic. If that's helpful, let me know and I'll supply it!

PiteousHonking, 2020-12-14:
Do you think it would be possible to flash them over serial without making them unusable? The plugs and bulbs from ultrabrite are a good pricepoint if they could be made to work with tasmota

kn100, 2020-12-21:
I mean "destructive" is a powerful word, they're still usable, just a bit ugly! I wrote this blog post recently going over the process for the plugs: https://kn100.me/declouding-chinese-wifi-plugs/

PiteousHonking, 2020-12-21:
That’s exactly what I was hoping for - now I know where the chip is that small area is easy to remove and refill. Thanks 😊

tomjmul, 2021-01-10:
Were you able to solder the FTDI connections without desoldering the ESP board? From looking at your pics, it seems that the main board has to be desoldered from the live and neutral pins, then the ESP desoldered to get to GPIO0.  Would love to be wrong

kn100, 2021-01-10:
Yes, I was. If you see the blog post I posted above, it'll become clearer the method I took to get at the pins.

wdmtech, 2021-05-21:
Also looking for some help with this plug! Tried all the mentioned steps, ended up following your blog @kn100 (thanks!) but still cant flash the chip. I've verified all my connections to make sure I haven't bodged the soldering, have tried all sorts from using an external PSU, resetting by holding the RST to low while entering flash mode with GPIO0 to ground, but never get an RX signal unless I use RST.

kn100, 2021-05-24:
@wdmtech are you sure you're serial adaptor is outputting 3.3v level digital signals? A lot of these cheaper serial adapters have both 5v and 3.3v power supply pins, but actually output their digital signals (on the rx/tx) pins at 5v. I had to replace mine for this reason.

wdmtech, 2021-05-26:
I've successfully flashed many ESP-01s using this FTDI adapter, but I'll try another one I have lying around just in case, thanks for the idea @kn100 !