ct-Open-Source/tuya-convert #803: Brilliant Dimmer Mechs no longer flashing

markrolandbooth, 2020-10-19:
Hi,

I have previously flashed 3x of the Brilliant Dimer Mechs without issue using Tuya-Convert on my Pi3.

I purchased three more of these mechs today, and they are not flashing when using the latest version of Tuya-convert

**Device Name:** Brilliant Dimmer Mech [link](https://www.brilliantsmart.com.au/smart-products/electrical/smart-mech/)
**Revision:** Item marked with "2020-23" printed on the side
**Tuya-Convert version:" 2.4.4
**Logs:** See attached

The item is put into flashing mode by pushing the mech 9 times in succession (mech led flashes quickly).

After the start_flash script is run, I join my mobile to the vtrush-flash SSID, and press enter.

At this point, the smart config pairing procedure starts, and it seems to register with the mech, as during the first run of "Resending SmartConfig packets" the light globe pulses momentarily.

But the resending attempts reach the 10x retry limit and end, without the device being flashed.

Logs are attached, thank you in advance.
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5398871/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5398872/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/5398873/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/5398874/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5398875/smarthack-wifi.log)

dmaunder, 2020-10-21:
Yeah I bought 8 of these a couple of months ago and have not been able to flash them either, same symptoms. The switch mechs bought at the same time all flashed fine.

kn100, 2020-10-21:
I'm not the maintainer or related to this repo but you're probably affected by https://github.com/ct-Open-Source/tuya-convert/wiki/Collaboration-document-for-PSK-Identity-02 - of which there is no solution yet except for hardware level flashing.

kn100, 2020-10-28:
I think you can pretty well establish this is the issue based on the behaviour you're experiencing. For me, once I put the plug in config mode (quick flashing) and tell Tuya-convert that I've done that, tuya-convert starts to emit the UDP packets for smartconfig. The plug then immediately stops flashing, but never connects to the vtrust-flash AP. In the PSK logs, you'll see loads of errors!

kn100, 2020-10-28:
I'd be curious to know just how inaccessible the IO0 pad is - you only need a temporary connection there to ground on boot to get it into flash mode!

markrolandbooth, 2020-10-29:
So the TYE2s module has a single circular IO0 pad on it's rear, and that whole face of the PCB is flat-soldered against another board (see pics attached). I can access the +3vdc/GND/RX/TX contacts, but not the IO0 one. Its pad is also not soldered to any corresponding pad on the board behind it (I sacrificed one of the units to discover this).

I don't suppose you're aware of any way to enter programming mode without IO0 (AT commands or the like)?

[image]
[image]

aSmig, 2020-12-03:
See #483. Confirmation is in your smarthack-psk.log where you have lines starting with "ID: 02".

Wow, that is an obnoxious placement of the TYWE2S module. For getting into bootloader mode without GPIO-0, any means that causes booting from flash to fail will revert to ROM. Tapping RST many times quickly with a ground wire has worked for me.

markrolandbooth, 2020-12-04:
> See #483. Confirmation is in your smarthack-psk.log where you have lines starting with "ID: 02".
>
> Wow, that is an obnoxious placement of the TYWE2S module. For getting into bootloader mode without [...]

Thank you aSmig! I will give it a try and see how I go.