ct-Open-Source/tuya-convert #806: Cannot flash Treatlife SS01 3-Way Switch

unmesh59, 2020-10-23:
Treatlife SS01 3-Way Switch, FCC ID 2ANIFSS01

I am using a RPi setup that I've used successfully other switches and plugs including the Treatlife SS02. This device has a template on blakaddr.com dated 2020-05-20 so something may have changed since then to prevent flashing with Tasmota. The fast blinking Red LED turns off within seconds of hitting ENTER on Tuya-Convert and then the script retries until it gives up

Logs attached

[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5426688/smarthack-psk.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5426689/smarthack-wifi.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/5426690/smarthack-udp.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5426691/smarthack-mqtt.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/5426693/smarthack-web.log)

unmesh59, 2020-10-23:
I stumbled across the link https://github.com/ct-Open-Source/tuya-convert/wiki/Collaboration-document-for-PSK-Identity-02 which says that PSK's beginning with 02 are not supported and specifically calls out the SS01 by name.

Once someone confirms that nothing has improved since that was last updated, I will close this issue.

Thanks

kueblc, 2020-10-23:
That's correct, sadly no workaround available for PSK 02 at this time.

closed 2020-10-23 by kueblc