ct-Open-Source/tuya-convert #807: Smartplug Gosund EP2 don't work - PSK ID Starts with 02

realhagie, 2020-10-23:
I ordered a 4 pack of gosund Smartplugs. They look like the famous SP111 but are a "newer revision". These are the new normal if you order them on amazon germany.

```
new client on port 443 from 10.42.42.16:50794
ID: 0242416f68626d643661473931494652316f4b4fb394ff2cba61d1b5f3daa88d4a8bdcf8584317699fbf761310c0f2870c
PSK: 9491448e78c3cb726d364f355d35f6033e34dd59bb2533dcfee105c4cdabf39b
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
```

Looks like they changed something and OTA flash isn't possible yet. 

Is here anyone skilled enough to inspect / examine / hack these plugs ? I'm happy send / donate one , especially within Germany, to someone.

SunsetGolfer, 2020-10-23:
New Chip :-(

Not supported any longer...

hannemann, 2020-10-24:
Do you know if its possible to flash them via Serial connection?

realhagie, 2020-10-24:
> Do you know if its possible to flash them via Serial connection?

Well right now I didn't even know how to open these little suckers .. from what I can see and what I feel there is no screw anymore in the PE hole at the bottom.

hannemann, 2020-10-24:
Yes, i think its hard to open them without doing any damage. Would be great if somenody knows if its flashable before giving it a try

jxtreme, 2020-10-24:
Can confirm, also got the 4-pack from Amazon.
The EP2 is not flashing via tuya-convert and quite sure affected by the PSK Bug (https://github.com/ct-Open-Source/tuya-convert/wiki/Collaboration-document-for-PSK-Identity-02).
Unfortuntely the plug is now glued (clear plastic to the base) and soldered (ground contacts to the base, with the screws hidden beneath the soldered ground bar).
Got it open almost undamaged and can confirm there is still an ESP (ESP8285) in there, but no obvious ISP pinout.

Detailed instruction to open this sucker (they really don't want anyone to open it anymore):
- carefully seperate the clear plastic from the white base (glued at the contact point) - i used a small saw and carefully split the parts, then pushed them apart with a screw driver. The clear inlet looks almost the same as in the SP111 plug and slides right out of the top.
- once the top is off, there's the nasty part: the bar for the ground contacts is soldered onto contact from the base that goes right trough the entire plug. You need to unsolder it to remove the ground bar and be able seperate top and bottom. Beneath the removed ground bar are two small screws holding the plug insert in place. Once removed you can push the entire insert out of the base by pushing the plug pins into the base.

kueblc, 2020-10-24:
Indeed PSK beginning with 02 is a known issue #483. More information on the wiki. No workaround available at this time.

You can try flashing by serial. Before opening, confirm that it is an ESP82xx by checking the MAC address, found in the wifi log or as part of the device ID. OUI should be Espressif Inc.

Closing as duplicate of #483, but feel free to continue discussion.

closed 2020-10-24 by kueblc

tckb, 2021-02-26:
Darn it! I just ordered thinking the exact same thing as @realhagie   @kueblc  how do I flash by serial? I am one step from returning it 👎

hannemann, 2021-03-04:
I managed to flash by serial all of my 4 devices following this guide: https://github.com/arendst/Tasmota/discussions/10350

I recommend to read through the whole thread before attempting to open one of the devices. It's kind of a hard thing to do without any damage and you will find some tips on how to do it. There is even an stl to replace the clear Plastic part with a 3d print  in case it breaks