ct-Open-Source/tuya-convert #811: Cannot build docker Tuya-convert

kylow-may, 2020-11-02:
Hi,

I am new to this world of coding and is learning all through what is discussed. Unless I did not get it, I have found no issue (apart from the alternate docker installation) on how to resolve this.

I tried to build docker tuya-convert and it is telling me that there is no docker image for my Raspberry Pi 4B. Below is an extract of my command and the error. How do I get a docker image built for my architecture? Thanks in advance.

> _pi@raspberrypi:~/tuya-convert $ docker build -t tuya:latest .
> Sending build context to Docker daemon  5.263MB
> Step 1/7 : FROM phusion/baseimage:master
> master: Pulling from phusion/baseimage
> no matching manifest for linux/arm/v7 in the manifest list entries_

alexyao2015, 2020-11-03:
Replace phusion/baseimage:master with debian

kylow-may, 2020-11-04:
Thanks. Managed to progress to downloading the images but it failed at Step 2/7 of the build process.
_

> pi@raspberrypi:~/tuya-convert $ docker build -t tuya:latest .                                                                                                                                 Sending build context to Docker daemon  5.263MB                                                                                                                                               Step 1/7 : FROM debian                                                                                                                                                                        latest: Pulling from library/debian                                                                                                                                                           5c0fdcca2cbb: Pull complete                                                                                                                                                                   Digest: sha256:8414aa82208bc4c2761dc149df67e25c6b8a9380e5d8c4e7b5c84ca2d04bb244                                                                                                               Status: Downloaded newer image for debian:latest                                                                                                                                               ---> a6d76f89ee04                                                                                                                                                                            Step 2/7 : RUN apt-get update && apt-get install -y sudo iproute2 iputils-ping                                                                                                                 ---> Running in ef4fb60247d4                                                                                                                                                                 failed to create endpoint inspiring_black on network bridge: failed to add the host (veth3577b5d) <=> sandbox (veth1ece32f) pair interfaces: operation not supported

_

alexyao2015, 2020-11-04:
Hmm I also initially struggled with the docker install. The current implementation seems less than ideal. 

I did a quick Google and it seems like you need to restart your pi.

diijkstra, 2020-11-26:
> Replace phusion/baseimage:master with debian

I would rather suggest using `phusion/baseimage:master-arm` or `phusion/baseimage:master-arm64`. It seems that `phusion/baseimage:master` is not packed for multiple archs and does not work on arm.

zeddski, 2020-12-18:
to help people out: 
* pico Dockerfile
* replace "phusion/baseimage:master" with "phusion/baseimage:master-arm" for 32-bit OS or "phusion/baseimage:master-arm64" for 64-bit OS.
* Ctrl-X
* y, Enter
* Try again Build again

alexyao2015, 2020-12-18:
I did manually create some images, no dockerfile, but it does work and is easy to use. No building required. If anyone is interested: https://hub.docker.com/r/yaoa/tuya-convert

hexates, 2020-12-21:
> > Replace phusion/baseimage:master with debian
> 
> I would rather suggest using `phusion/baseimage:master-arm` or `phusion/baseimage:master-arm64`. It seems that `phusion/baseimage:master` is not packed for multiple archs and does not work on arm.

This seems to be working for me so far, thank you so much!

hexates, 2020-12-21:
> Replace phusion/baseimage:master with debian

This was what finally worked for my raspberry pi 3b+!