ct-Open-Source/tuya-convert #812: Device did not appear with the intermediate firmware

RoelDU, 2020-11-04:
Im trying to flash the Wireless Smart Valve Model SM-AW713. 
I followed a YouTube video where exactly the same valve was being flashed (called "Tasmotizing the Useelink Tuya Based Water Shutoff Valve").

LOGS:

smarthack-web.log:

Listening on 10.42.42.1:80
[32m[I 201104 10:19:01 web:2243][m 302 GET /generate_204 (10.42.42.26) 6.66ms
[32m[I 201104 10:19:01 web:2243][m 302 GET /generate_204_e74abc12-3c2e-47c6-8ba2-ada7ed1b864e (10.42.42.26) 4.49ms
[32m[I 201104 10:19:01 web:2243][m 302 GET /generate_204_110653cd-7aff-4e4f-b9f8-3d0b174af645 (10.42.42.26) 3.57ms
[32m[I 201104 10:19:01 web:2243][m 302 GET /generate_204_094fe3db-662d-4006-a3b9-6d87a1dc1c92 (10.42.42.26) 3.10ms
[32m[I 201104 10:19:01 web:2243][m 302 GET /generate_204_7163b615-3f55-407f-8f9a-320452721b2a (10.42.42.26) 4.01ms
[32m[I 201104 10:19:02 web:2243][m 302 GET /generate_204_110653cd-7aff-4e4f-b9f8-3d0b174af645 (10.42.42.26) 2.90ms
[32m[I 201104 10:19:02 web:2243][m 200 GET / (10.42.42.26) 4.45ms
[32m[I 201104 10:19:02 web:2243][m 302 GET /favicon.ico (10.42.42.26) 2.52ms
[32m[I 201104 10:19:02 web:2243][m 304 GET / (10.42.42.26) 2.78ms
[32m[I 201104 10:19:03 web:2243][m 302 GET /generate_204_110653cd-7aff-4e4f-b9f8-3d0b174af645 (10.42.42.26) 3.00ms
[32m[I 201104 10:19:03 web:2243][m 302 GET /generate_204_110653cd-7aff-4e4f-b9f8-3d0b174af645 (10.42.42.26) 2.47ms
[32m[I 201104 10:19:04 web:2243][m 302 GET /generate_204_55122ac0-298b-4be6-b7a5-d784b80c4064 (10.42.42.26) 3.46ms
[32m[I 201104 10:19:14 web:2243][m 302 GET /generate_204_068b1160-f96c-46cf-acca-5455980210e1 (10.42.42.26) 3.28ms
[32m[I 201104 10:19:17 web:2243][m 302 GET /generate_204_57f8a2e8-cb85-4334-a2e4-7aa3eb1159c3 (10.42.42.26) 4.60ms
[32m[I 201104 10:19:33 web:2243][m 302 GET /generate_204_67fbbe90-47b6-46e3-90c8-e0a55b73442d (10.42.42.26) 3.49ms
[32m[I 201104 10:19:34 web:2243][m 302 GET /generate_204_856a7057-ad90-44c4-b9a8-3682d16b4425 (10.42.42.26) 3.38ms
[32m[I 201104 10:19:48 web:2243][m 302 GET /generate_204_ae5aa231-2fdf-40be-aeee-1ee033771035 (10.42.42.26) 4.07ms
[32m[I 201104 10:19:57 web:2243][m 302 GET /generate_204_c633e534-a569-41f0-bbd0-b8f0d4cd9524 (10.42.42.26) 3.32ms
[32m[I 201104 10:19:57 web:2243][m 302 GET /generate_204_9efc48cd-2c7a-4f3e-87a9-be8ad44d34f7 (10.42.42.26) 3.29ms
[32m[I 201104 10:19:57 web:2243][m 302 GET /mobile/status.php (10.42.42.26) 2.92ms
[32m[I 201104 10:19:57 web:2243][m 200 GET / (10.42.42.26) 3.00ms
[32m[I 201104 10:20:15 web:2243][m 302 GET /generate_204_1e098610-dde9-472f-958f-37e6e7227bd5 (10.42.42.26) 3.54ms
[32m[I 201104 10:21:35 web:2243][m 302 GET /generate_204_c5a49581-3318-413b-9537-07e183daa176 (10.42.42.26) 3.47ms
^CReceived SIGINT, exiting...

smarthack-wifi.log:

System info
===========
94acc3b
Linux raspberrypi 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux
OpenSSL 1.1.1d  10 Sep 2019
Dnsmasq version 2.80  Copyright (c) 2000-2018 Simon Kelley
Compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile

This software comes with ABSOLUTELY NO WARRANTY.
Dnsmasq is free software, and you are welcome to redistribute it
under the terms of the GNU General Public License, version 2 or 3.
hostapd v2.8-devel
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2019, Jouni Malinen <j@w1.fi> and contributors
Python 3.7.3
===========
Attempting to stop wpa_supplicant
Configuring AP interface...
Error: Device for nexthop is not up.
Error: Device for nexthop is not up.
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:f6:16:98 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
wlan0: AP-STA-CONNECTED 84:f3:eb:ea:a3:3b
wlan0: AP-STA-CONNECTED 74:60:fa:44:f8:ea
wlan0: AP-STA-DISCONNECTED 84:f3:eb:ea:a3:3b
wlan0: AP-STA-CONNECTED 84:f3:eb:ea:a3:3b
wlan0: AP-STA-DISCONNECTED 84:f3:eb:ea:a3:3b
wlan0: AP-STA-CONNECTED 84:f3:eb:ea:a3:3b
wlan0: AP-STA-DISCONNECTED 84:f3:eb:ea:a3:3b
wlan0: AP-STA-CONNECTED 84:f3:eb:ea:a3:3b
wlan0: AP-STA-DISCONNECTED 84:f3:eb:ea:a3:3b
wlan0: AP-STA-CONNECTED 84:f3:eb:ea:a3:3b
wlan0: AP-STA-DISCONNECTED 84:f3:eb:ea:a3:3b
wlan0: interface state ENABLED->DISABLED
wlan0: AP-STA-DISCONNECTED 74:60:fa:44:f8:ea
wlan0: AP-DISABLED 
wlan0: CTRL-EVENT-TERMINATING 
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...

smarthack-mqtt.log

smarthack-psk.log:

could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.23:31453
ID: 0242416f68626d643661473931494652318e4e96994858faa993c9cc2b8e9a3c3f08d08f5e7209beeb364c0dabba22a591
PSK: df29c3b87de1778a41f0ee86e0ec067031ad4b2d28f38c8b8a92d530eca19243
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.26:45688
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.26:45692
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.23:48779
ID: 0242416f68626d643661473931494652318e4e96994858faa993c9cc2b8e9a3c3f08d08f5e7209beeb364c0dabba22a591
PSK: df29c3b87de1778a41f0ee86e0ec067031ad4b2d28f38c8b8a92d530eca19243
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.23:19271
ID: 0242416f68626d643661473931494652318e4e96994858faa993c9cc2b8e9a3c3f08d08f5e7209beeb364c0dabba22a591
PSK: df29c3b87de1778a41f0ee86e0ec067031ad4b2d28f38c8b8a92d530eca19243
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.26:45696
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!

Getting the following error:

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fa                                                               st). This is usually done by pressing and holding the primary button of the devi                                                               ce
   Make sure nothing else is plugged into your IoT device while attempting to fl                                                               ash.
3. Press ENTER to continue
======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
..............
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N] Device did not appear with the                                                                intermediate firmware
======================================================
Cleaning up...
Closing AP
Exiting...
pi@raspberrypi:~/tuya-convert $ evice did not appear with the intermediate firmwareDevice did not appear with the intermediate firmwareDevice did not appear with the intermediate firmwareDevice did not appear with the intermediate firmware

jonadis, 2020-11-04:
[https://github.com/ct-Open-Source/tuya-convert/wiki/Collaboration-document-for-PSK-Identity-02](url)

Crag-Monkey, 2020-12-30:
I've seen the same thing with most of the devices I've tried to flash (only ever gotten one to work, actually). I _suspect_ that the firmware has been patched to prevent Tuya-Convert from connecting to these newer devices.

kueblc, 2020-12-30:
Oops must have missed this issue. @jonadis and @Crag-Monkey are right, this is a duplicate of pinned issue #483.

closed 2020-12-30 by kueblc