ct-Open-Source/tuya-convert #814: Feit BR30 does not flash

bani6809, 2020-11-08:
Feit BR30 do not appear to be flashable. https://www.amazon.com/gp/product/B07GFV1Y9C

Put the bulb in fast pairing mode. As soon as the packets are sent, the bulb stops flashing. But the bulb never sends any requests.

This seems completely unrelated to the issue 483 PSK issue. No PSK involved because no traffic.

root@raspberrypi:/home/pi/tuya-convert# ./start_flash.sh
tuya-convert v2.4.4
Checking for network interface wlan0... Found.
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
..............
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N] n
======================================================
Cleaning up...
Closing AP
Exiting...

::::::::::::::
smarthack-mqtt.log
::::::::::::::
1604824584: mosquitto version 1.5.7 starting
1604824584: Using default config.
1604824584: Opening ipv4 listen socket on port 1883.
1604824584: Opening ipv6 listen socket on port 1883.
^C1604824768: mosquitto version 1.5.7 terminating
::::::::::::::
smarthack-psk.log
::::::::::::::
new client on port 443 from 10.42.42.13:2530
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.13:2531
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.13:2533
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.13:2534
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.13:2535
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.13:2537
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.13:2538
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
^CTraceback (most recent call last):
  File "./psk-frontend.py", line 113, in <module>
    main()
  File "./psk-frontend.py", line 106, in main
    r,_,_ =  select.select(readables, [], [])
KeyboardInterrupt
::::::::::::::
smarthack-udp.log
::::::::::::::
Listening for Tuya broadcast on UDP 6666
Listening for encrypted Tuya broadcast on UDP 6667
^C
::::::::::::::
smarthack-web.log
::::::::::::::
Listening on 10.42.42.1:80
[I 201108 08:37:08 web:2243] 302 GET /success.txt?ipv4 (10.42.42.13) 3.28ms
[I 201108 08:37:08 web:2243] 302 GET /success.txt (10.42.42.13) 1.70ms
[I 201108 08:37:08 web:2243] 200 GET /?ipv4= (10.42.42.13) 3.18ms
[I 201108 08:37:08 web:2243] 302 GET /connecttest.txt (10.42.42.13) 2.53ms
[I 201108 08:37:09 web:2243] 302 GET /connecttest.txt (10.42.42.13) 1.83ms
[I 201108 08:37:11 web:2243] 302 GET /success.txt (10.42.42.13) 1.48ms
[I 201108 08:37:14 web:2243] 302 GET /success.txt (10.42.42.13) 1.56ms
^CReceived SIGINT, exiting...
::::::::::::::
smarthack-wifi.log
::::::::::::::
System info
===========
94acc3b
Linux raspberrypi 5.4.72-v7l+ #1356 SMP Thu Oct 22 13:57:51 BST 2020 armv7l GNU/Linux
OpenSSL 1.1.1d  10 Sep 2019
Dnsmasq version 2.80  Copyright (c) 2000-2018 Simon Kelley
Compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile

This software comes with ABSOLUTELY NO WARRANTY.
Dnsmasq is free software, and you are welcome to redistribute it
under the terms of the GNU General Public License, version 2 or 3.
hostapd v2.8-devel
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2019, Jouni Malinen <j@w1.fi> and contributors
Python 3.7.3
===========
Configuring AP interface...
RTNETLINK answers: File exists
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr dc:a6:32:12:ad:e1 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
wlan0: AP-STA-CONNECTED a0:20:a6:16:f7:04
wlan0: AP-STA-CONNECTED 18:69:d8:53:ab:36
wlan0: AP-STA-CONNECTED ac:d1:b8:c0:c2:91
wlan0: AP-STA-DISCONNECTED 18:69:d8:53:ab:36
wlan0: AP-STA-CONNECTED 18:69:d8:53:ab:36
wlan0: AP-STA-DISCONNECTED 18:69:d8:53:ab:36
wlan0: AP-STA-CONNECTED 18:69:d8:53:ab:36
wlan0: AP-STA-DISCONNECTED ac:d1:b8:c0:c2:91
wlan0: interface state ENABLED->DISABLED
wlan0: AP-STA-DISCONNECTED 18:69:d8:53:ab:36
wlan0: AP-STA-DISCONNECTED a0:20:a6:16:f7:04
wlan0: AP-DISABLED
wlan0: CTRL-EVENT-TERMINATING
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...

bani6809, 2020-11-08:
`
[log.txt](https://github.com/ct-Open-Source/tuya-convert/files/5506044/log.txt)
`

bani6809, 2020-11-08:
Looks like related to #784 

So a new class of OTA upgrade failures not related to PSK.

kueblc, 2020-11-08:
I see three devices in your wifi log, can you identify them?

bani6809, 2020-11-09:
ac:d1:b8:c0:c2:91  laptop
a0:20:a6:16:f7:04  makerfocus d1 mini
18:69:d8:53:ab:36  appears to be the feit br30, shows up when i put the bulb into fast pairing

kueblc, 2020-11-10:
All ESP82xx have an OUI of Espressif Inc. Unfortunately your Feit BR30 is not using an ESP82xx and thus is not capable of running third party ESP firmware. Sorry to be the bearer of bad news!

closed 2020-11-10 by kueblc

bani6809, 2020-11-10:
disassembled the bulb, looks like feit is using the new arm-based tuya wb2l module. the FCC ID is the same as old feit br30 bulbs with esp8266, this seems to be an fcc violation.

nsh99, 2020-11-10:
I purchased the same bulb from Lowes. Same beginning MAC address: https://maclookup.app/macaddress/1869D8