ct-Open-Source/tuya-convert #815: Failed to run on Raspberry pi 4b

krazyjakee, 2020-11-08:
Hi there,

I can't seem to run tuya-convert on raspberry pi 4b.
I'm running it from hassio which I believe is based on alpine?

When running i get...

```
AP mode not supported
Please attach a wifi card that supports AP mode.
```

bani6809, 2020-11-10:
tuya-convert works for me on a pi 4b, using the 2020-08-20-raspios-buster-armhf-lite image.

becmar39, 2020-12-02:
Are you using OnBoard Wifi Card? 
Not all USB Wifi Cards supporting AP Mode.

Lebo-Wang, 2020-12-10:
It works for me. FYI, for my gosund device, if it shows that 'Could not fetch a complete backup
Do you want to continue anyway?'. Plz choose YES. I stopped and probably need to broke it for a wired flash. LOL

kueblc, 2020-12-10:
@Lebo-Wang your device is not broken, it is simply running the intermediate firmware used by tuya-convert to flash your firmware. You can restart the tuya-convert process to continue, at which point you can choose to return to stock or flash 3rd party firmware.

Lebo-Wang, 2020-12-10:
> @Lebo-Wang your device is not broken, it is simply running the intermediate firmware used by tuya-convert to flash your firmware. You can restart the tuya-convert process to continue, at which point you can choose to return to stock or flash 3rd party firmware.

Good to know. Let me try again to continue at the stucked point. Although, it did not response to tuya-convert, or any button press. Thank you!

Lebo-Wang, 2020-12-10:
> @Lebo-Wang your device is not broken, it is simply running the intermediate firmware used by tuya-convert to flash your firmware. You can restart the tuya-convert process to continue, at which point you can choose to return to stock or flash 3rd party firmware.

Tried many times, it always showed as 'Could not reach the device!'. I guess it is out of power for some reason?

Lebo-Wang, 2020-12-10:
> @Lebo-Wang your device is not broken, it is simply running the intermediate firmware used by tuya-convert to flash your firmware. You can restart the tuya-convert process to continue, at which point you can choose to return to stock or flash 3rd party firmware.

Finally, it works! Many thanks again!