ct-Open-Source/tuya-convert #816: BSD33 10A Plug

jaime-rubio, 2020-11-10:
Hi @kueblc 

I had already flashed other devices with tuya-convert 2 years ago, but now.... no way.
I've tried with a simple smart plug BSD33 (originial firmware 1.4 MCU).
Everything is ok except that the device is not flashed. It stucks at smart config

**MQTT:**
1605026323: mosquitto version 1.4.15 (build date Tue, 18 Jun 2019 11:42:22 -0300) starting
1605026323: Using default config.
1605026323: Opening ipv4 listen socket on port 1883.
1605026323: Opening ipv6 listen socket on port 1883.

**WiFi: (I have masked the MAC on this topic)**
Using interface wlx0013f7aa089f with hwaddr XX:XX:XX:XX:XX:XX and ssid "vtrust-flash"
wlx0013f7aa089f: interface state COUNTRY_UPDATE->ENABLED
wlx0013f7aa089f: AP-ENABLED 
wlx0013f7aa089f: AP-STA-CONNECTED XX:XX:XX:XX:XX:XX

**PSK:**
new client on port 443 from 10.42.42.13:54620
new client on port 443 from 10.42.42.13:54784

**UDP:**
Listening for encrypted Tuya broadcast on UDP 6667
10.42.42.15 {'ip': '10.42.42.15', 'gwId': 'bfe5dcad0be57ad07aiphr', 'active': 0, 'ablilty': 0, 'encrypt': True, 'productKey': 'zmuquaytzuth4gkp', 'version': '3.3'}

**Web:**
Listening on 10.42.42.1:80
[32m[I 201110 17:38:45 web:2243][m 302 GET /generate_204 (10.42.42.13) 0.75ms
[32m[I 201110 17:38:55 web:2243][m 302 GET /generate_204 (10.42.42.13) 0.58ms
[32m[I 201110 17:38:57 web:2243][m 302 GET /generate_204 (10.42.42.13) 0.56ms

And here... it hangs !
Is this still blocking because the certificate validation? has anyone succeeded with a recent firmware?
I have no modified the orignal productKey. Should I?

Thank you

kueblc, 2020-11-10:
It looks like this device may no longer be using an ESP82xx microchip. You can verify this by checking the MAC address for an OUI of "Espressif Inc". If it is not "Espressif Inc" it is not an ESP and thus can't run ESP firmware.

kotzer3, 2021-01-18:
hi there, having same issue, but there is an Espressif Inc module in. Mac starts with `70:03:9f` --> `70:03:9F Espressif Inc.`
I am bringing the device into setup mode, by pressing the button till it blinks blue and red (fast mode)

The only entries which are generated by tuya-convert:
**web:**
```
302 GET /generate_204 (10.42.42.35) 4.74ms
405 POST /rqd/async?aid=<some-fancy-guid> (10.42.42.35) 26.49ms
405 POST /rqd/async?aid=<some-other-fancy-guid> (10.42.42.35) 7.13ms
```

**psk:**
```
new cient on port 443 from 10.42.42.23:30606
ID: <long hash with hex chars>
PSK: <another different long hash, half length of ID>
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
<repeated many times>

new client on port 443 from 10.42.42.35:41014
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is propably just your phone!
new client on port 443 from 10.42.42.35:41016
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is propably just your phone!
new client on port 443 from 10.42.42.35:41018
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is propably just your phone!
```

from log is looks like the mobile generates the first block and the plug generates the part, which is marked as: propably your phone..

behaviour while trying to convert to away from tuya..
- raspi booted up, started the start_flash.sh
- mobile logged into wifi
- bringing plug into pairmode by pressing the button for ~ 10 secs until it blinks red/blue
- pressing ENTER (after confirming that already started processes have to been canceled)
- then during sending configuration the plugs relais is toggled (click - off - led is off), then 500mseccs later (click - both leds red and blue are on, it displays purple)
- bringing plug back into pair-mode and the process is repeated:

logs on rpi-screen during pairing-process: 
```
 Starting smart config....
 Sending SSID		vtrush-flash
 Sending wifiPassword
 Sending token		00000000
 Sending secret		0101
 ........................
 SmartConfig complete.
 Resending SmartConfig Packets
 ........................
 SmartConfig complete.
 Resending SmartConfig Packets
 ........................
 SmartConfig complete.
 Resending SmartConfig Packets
 ........................
 SmartConfig complete.
 Resending SmartConfig Packets
 ........................
 SmartConfig complete.
 Resending SmartConfig Packets
 ........................
 SmartConfig complete.
 Resending SmartConfig Packets
  ........................
 SmartConfig complete.
 Resending SmartConfig Packets
 ........................
 SmartConfig complete.
 Resending SmartConfig Packets
 ........................
 SmartConfig complete.
 Resending SmartConfig Packets
 ........................
 SmartConfig complete.
 Resending SmartConfig Packets
 ........................
 Receive did not appear with the intermediate firmware
 Check the ....
```
 
 I have two of the plugs, with both of them same behaviour.
 did already pull latest version from git. (this morning again)
 
 I did not connect these before via the tuya smart-life app. Maybe this is important?
 Any hints/suggentions?

bog500, 2021-01-18:
Does your ID starts with 02 ?

Check out https://github.com/ct-Open-Source/tuya-convert/issues/483

kotzer3, 2021-01-21:
totally right, it starts with 024... like all the others (also bought on dec 2020.. on aliexpress, but its a vsd33, 16A plug. no matter..)
so just wait for a solution or open the plug and flash with wires connected (temporary soldered)...