ct-Open-Source/tuya-convert #817: LEDLite (Tuya branded) LTGU4WIFI bulb.

danielbayley80, 2020-11-12:
I'm trying to flash a LEDLite (Tuya branded) LTGU4WIFI bulb.

I have tuya convert running and I can connect with my mobile device.
I *think* the bulb is able to hit the wifi AP because it stops flashing when I have it running.
But, it always times out with "Device did not appear with the intermediate firmware".
I've tried cycling the bulb multiple times to see if it retries and with no result.

I see these errors in the log :

new client on port 443 from 10.42.42.33:26523
ID: 0242416f68626d64366147393149465231cd13dde5bb4f79c219fee9923f16d3d91b6b3574fc54ba3e461c22357b9e953a
PSK: aca251fc7bd802c9401028997d2ea40360af0de637a4850571bad85ca76a0d34
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1123)

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5527799/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5527800/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/5527801/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/5527802/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5527803/smarthack-wifi.log)

Noschvie, 2020-11-12:
See #483.

danielbayley80, 2020-11-12:
thanks @Noschvie . I see there is a wiki page on this issue : https://github.com/ct-Open-Source/tuya-convert/wiki/Collaboration-document-for-PSK-Identity-02

I'm not sure what jurisdictions people are in but I've noticed that Tuya devices have flooded the UK market.
There are some really big name retailers pushing these devices and their customers are likely unwittingly giving up their data to 3rd parties. some of the privacy policies aren't exactly clear who is behind the service.

Not only that they are putting the longevity and maintenance of products in the hands of third parties who on a whim might change things, or be sold on. Also I read they stop working if you cut off their "phone home".

Tuya is also a chinese company correct?

I think as well as the technical angle we (as a community) should push the regulatory angle. 

There are some very big names selling these products and the reputational damage would be huge if an issue emerged. I don't know how much diligence they do on these things but one electrical wholesaler I spoke to was very naïve and referred me to tuya's response which was essentially "don't worry about it we're a big company and for the price what do you expect" (paraphrasing).

I am drawing up a list and doing some further investigations. I've made a number of test purchases and will write up my findings and send to industry experts, ICO (our data regulator), etc.

this market needs regulating in a way that protects the rights/ freedoms / privacy of consumers without stifling innovation. 

Another option would be source these sort of products directly from the manufacturer pre-flashed or at least ready to adopt firmware. Is the enthusiast market big enough?

nsh99, 2020-11-12:
There is at least one company pre-flashing Tasmota on devices now:  https://templates.blakadder.com/preflashed.html
I would love to see a US supplier take up the helm but I think the market for enthusiasts is fairly small overall.  Most people just want it to work and don't think about the network traffic/data flow.

danielbayley80, 2020-11-12:
awesome thanks. 

I've started to make a list of big name retailers and the brands they are selling. most look like they lead back to tuya. I've sent them emails about their privacy and security. it will be interesting to see what they say. 

People might pay more attention if I make a bit of a fuss and get a tech journalist involved.

kueblc, 2020-11-15:
Hi Dan,

Yes you are experiencing #483, a known issue with no resolution at this time.

> People might pay more attention if I make a bit of a fuss and get a tech journalist involved.

This project was started by tech journalists, in fact, this repository is hosted by c't. There have been multiple articles which are linked in the README.

Not to discourage you, but the suggestions you make are essentially what we have been doing for the past two years. It does not seem like Tuya wants to play ball. Review the README for more information.

Nevertheless, more pressure on existing manufacturers is good. I have also been tossing around the idea of making a bulk direct order to have devices pre-flashed like you mentioned, but I need to know I can sell and distribute them before I invest that much money.

Closing this issue as duplicate of #483, you can watch that thread for future announcements, but feel free to continue using this thread for discussion.

closed 2020-11-15 by kueblc

danielbayley80, 2020-11-16:
Thanks @kueblc. I'll update you as and when I get anywhere.

> Hi Dan,
>
> Yes you are experiencing #483, a known issue with no resolution at this time.
>
>
> This project was started by tech journalists, in fact, this repository is hosted by c't. There have been [...]