ct-Open-Source/tuya-convert #818: PSA: fix for "Please attach a Wifi card that supports AP Mode"

Ltek, 2020-11-14:
THis is in hope someone else has this problem and finds this since after reading about 20 articles / how-tos not a single one mentions needing SUDO to run this!

If your WLAN0 reports as the wifi card using IFCONFIG and is also what is set in tuya-convert/config.txt you prob have a perms problem.

run...

SUDO ./start_flash.sh

kueblc, 2020-12-12:
`sudo` should not be used when calling `start_flash.sh`. `sudo` is already called by the scripts when and only when necessary.

If the AP set up process failed for you, we should attempt to determine the cause rather than forcing the process to run as root.

grantslape, 2021-01-11:
For me, there was a more pertinent error message immediately above this that shined more light onto the issue.

In my case the script could not find `iw`. This was an issue of `iw` being installed as root.  Shimming `$PATH` resolved my issue:

```
export PATH="$PATH:/sbin"
```