ct-Open-Source/tuya-convert #819: Flashing Feit Dual Outlet not working

Ltek, 2020-11-15:
I'm new to all of this so maybe I'm missing something key...

Tuya-Convert running on Kali Live USB
Did all steps in guide to update all the packages.

SUDO ./start_flash.sh
- in a few seconds it shows 'SmartConfig complete' but quickly "retries" multiple times.
- Blue light on Feit continues to flash the entire time, through multiple 'retries'
- back-up of existing FW never shows or tries.
- at the end, says no intermediate firmware.

ideas?

thx for the help!

https://templates.blakadder.com/feit_electric-PLUG-WIFI-WP.html

eboston, 2020-11-15:
There is a new version of Tuya out that doesn't work with Tuya-Convert.  I have run into this with two different switches.  So until a new version of Tuya-Convert comes out, you might be out of luck.

Ltek, 2020-11-15:
How do we tell if the switch will work?

kueblc, 2020-11-15:
Please include your logs when opening a new issue in order for us to identify the problem.

eboston, 2020-11-15:
> How do we tell if the switch will work?

Only way I know of is to try flashing one.  Otherwise, I would say you have to assume that all Tuya devices currently have the latest unflashable version.

Ltek, 2020-11-16:
> Please include your logs when opening a new issue in order for us to identify the problem.
Logs attached.

FYI, read your Troubleshooting page - didnt see mention of where the log files are stored. 
Also

 good info from some random person on how to monitor the logs...
https://www.ozsmartthings.com.au/blogs/news/tech-guide-how-to-flash-tuya-wifi-switches-with-tasmota-with-tuya-convert

[drive-download-20201116T223531Z-001.zip](https://github.com/ct-Open-Source/tuya-convert/files/5550117/drive-download-20201116T223531Z-001.zip)

Ltek, 2020-11-19:
@kueblc  do the logs help?

kueblc, 2020-11-19:
Hi @Ltek 

Just got a chance to look at your logs. In your wifi log you have four devices connecting:
```
18:69:d8:c8:0d:7c
18:69:d8:c8:3a:1a
5e:b6:2e:ab:ec:98
94:bf:2d:c5:ee:54
```

We can only flash devices which use the ESP82xx microcontroller. You can identify the device based on the OUI, the first half of the MAC address. Unfortunately none of these are ESP devices, so you won't be able to install ESP firmware on these, OTA or not.

closed 2020-11-19 by kueblc

Ltek, 2020-11-20:
@kueblc thats odd, I thought the Feit Outlets used the ESP82xx ... it was on the compatibility page until 2 days ago, now that is gone.

kueblc, 2020-11-20:
Manufacturers have been silently switching to other chipsets for over a year now (#269)