ct-Open-Source/tuya-convert #824: Anyone got a RGBW downlight brand that consistently flashes?

pashdown, 2020-11-21:
Zemismart was flashing up until recently, now they're not.  I've tried three other brands on Amazon, and they're all blocking the flash.  Anyone have any recommendations at this point?  Tearing these beasts apart requires nearly destroying them, and I'd rather find something that works with tuya-convert.

Grimal, 2021-04-17:
I was looking too, didn't find any.  I resigned myself to buying the available units and if necessary swapping in ESP8266's.  So, I placed an order and received a half dozen (so far) from China from an Ali vendor named FrankEver.  They arrived with a Beken BK7231T on board, so a swap it will likely be.  I'd love to figure out building firmware and flashing these if I could ever find the time, because they look like a great little chip.  ARM based, 32 bit, fast, lots of features.  The lights themselves are nice, the frosted lens pops right out with a suction cup, and the high voltage side sits under a removable cap on the back.  Very easy to get into.

[image]

[image]

al3x1337, 2021-04-17:
> I was looking too, didn't find any. I resigned myself to buying the available units and if necessary swapping in ESP8266's. So, I placed an order and received a half dozen (so far) from China from [...]

Just going to quickly chime in here! Preface: I work for a wholesale company and we got samples of this product ordered. Not sure if this is on aliexpress. 

I've de soldered with a heat gun and flashed this [Ssluce](https://www.ssluce.com/product-item/6-pearl-series-slim-downlight-wifi-rgbcw/) light. Probably took 15/20 minutes of time but I was able to successfully convert it with used the [Deta ](https://templates.blakadder.com/deta_DET902HA.html) template. If i recall correctly I think only the TX pin was missing hence me having to desolder the TYWE2L chip. 

I'm very happy I was able to get a downlight with Tasmota and might look into ordering more in the future. Currently though my a fan and waiting for development on any Wifi/Bluetooth Mesh (light to light) chips that come out, similar to Yeelight with their Bluetooth Gateway + M2 Downlight.

https://cdn.discordapp.com/attachments/530862923363516417/817840106823090216/image0.jpg
https://cdn.discordapp.com/attachments/530862923363516417/816721674413801512/image0.jpg

Grimal, 2021-04-19:
I pulled the [https://github.com/bekencorp/bk7231_freertos_sdk](bk7231_freertos_sdk) from BekenCorp, but they utilized the commercial REVA RivieraWaves WiFi IP so I can't compile it.  So unfortunately it looks like a dead end for this SoC.

So, I popped the SoC off the board and will strain my eyes to their max stitching in some ESP8266 ESP-12 modules I've got laying around.  I'll be giving up hardware PWM which is a bummer, but they should work for my purposes.