ct-Open-Source/tuya-convert #825: Can’t flash Ayocor us3 with psk I’d 01

craigueh, 2020-11-22:
### Before opening a new issue, please read the following

**Are you reporting a successful flash?**
_Please add it to the wiki instead!_

**Do you have a question about configuring your newly flashed device?**
_Please ask in your alternative firmware's support channel instead!_

**Did you have issues getting a device to flash?**
_Please be sure to include your device **name, model, and firmware version** as well as your **logs**!_

**Please remember to close your issue when the problem has been addressed!**

kueblc, 2020-12-02:
> After a lot of attempts I managed to flash one to tasmota bit after a reboot it’s now dead. Does not connect to Wifi and does not respond to putting it back in flash mode.

Did you configure Tasmota to use your wifi? It would not connect to your wifi if you did not. If you did, you may have mistyped some of the information, in which case Tasmota has a recovery option, see the Tasmota documentation.

It would not go into "flash mode" if you installed Tasmota. That is a feature of the Tuya firmware that you overwrote.

Also did you try power cycling the device? Completely remove the device from power and wait a few moments before reconnecting it. Some devices have trouble booting new firmware until a full reset.

> Any ideas where to start with troubleshooting?

Upload your logs if you believe this is an issue with tuya-convert. Otherwise seek support from Tasmota.

craigueh, 2020-12-02:
Thanks for the tips but I had been through all that. I’ve over 50 esp devices so well used to the quirks of tasmota and esphome. 

I cracked open the plug and hooked up a serial adapter. It was spitting out a load of boot errors and resetting continuously so I flashed it again via serial and it booted tasmota but did the same again once i unplugged it. So I flashed it with esphome and it did a similar thing but could see watchdog timer reboots in esphome and flash errors. I flashed it again with a blank 1MB bin then a final reflash with esphome. This time it came up fine and is working fine ever since. 
Something obviously went weird with the initial flash. 

So I cracked open the other 3 U3s plugs that were failing to come up with intermediate firmware during tuya-convert and flashed them all by serial with esphome. 
Guess what, they all failed to boot the first time. After a flash with the 1MB blank bin file and a reflash with esphome they all come up and function normally. 
I’ve never seen this happen before with any device but it is strange. 

I can only assume that something has changed with the stock firmware on these devices that causes the initial flash to fail to boot. This is why even though these devices have the older psk 01 format they still don’t boot successfully with the intermediate firmware and so the tuya-convert process fails.

ttr, 2020-12-03:
@craigueh It seems I have same issue - id reported to be 01, connects to ap, sends request to gw.json and instantly exit EZ mode.
So what you're saying is that only way to flash those would be serial cable ? Asking as If this is a case, I will return them as not 'in a mood' to de-assembly those.
Thanks.

craigueh, 2020-12-03:
It would seem so. I tried for days with every possible option I could find on the net. Dev versions of tuya-convert, esp donor device, windows based device connecting to vtrust Wifi ap etc. Nothing worked so I gave up and had all 4 flashed within an hour by serial.

ttr, 2020-12-04:
@craigueh I know this will be OT so apologies for this, but i've decided to open plug and... can You tell me where are serial connectors on a board? can't find them nor anything over internet in this regard.
Thanks.

craigueh, 2020-12-04:
Here you go:
https://www.youtube.com/watch?v=7vjpcEedEBI

This guy has a page with the pin outs written on it in the video. The tx and rx connections actually have pads broken out which make them quite easy to solder. The vcc pin easy enough and you can find gnd in a bunch of places on the board. Gpio0 is easy as you can just touch the end of a dupont wire off the solder pad to enter boot mode. 

I didn’t desolder the circuit board from the mains pegs, it will gently bend back giving you just enough room for soldering. You can bend the board back down and glue the casing back in place leaving the whole job very neat.