ct-Open-Source/tuya-convert #826: Please document #483 PSK issue in toplevel readme

jackjansen, 2020-11-22:
For newcomers to tuba-convert (like me) the PSK issue (#483, https://github.com/ct-Open-Source/tuya-convert/wiki/Collaboration-document-for-PSK-Identity-02) is completely unknown, and it is not referenced at all in the toplevel readme.

So probably everyone who has recently come across this repo has gone through the same price as me, having to find out trough reading logs and such what might be the problem, then report an issue, then find that there are related issues, then finding out that this has been a known problem already for over a year.

I would suggest adding a section to the README.md, because there's a very good chance that recently purchased Tuya hardware has the problem.

doofy, 2020-11-23:
FYI: there are two pinned issues in the issue tracker. the wiki page is somewhat unreadable as to be not well formatted/structured

imduffy15, 2020-11-30:
💯  Would be great to increase visibility around this and create more noise

kueblc, 2020-12-02:
Great idea @jackjansen, I've added a few lines on this in #844. I've also been planning on adding detection for this condition to the script itself, it has fallen a little behind on my TODO list.

> FYI: there are two pinned issues in the issue tracker. the wiki page is somewhat unreadable as to be not well formatted/structured

Please feel free to help!

closed 2020-12-02 by kueblc