ct-Open-Source/tuya-convert #827: How can I contribute to PSK reverse engineering?

jackjansen, 2020-11-22:
I would like to contribute (time permitting) to the reverse engineering of the #483 PSK issue, but at the moment it is unclear how i can do so.

The issue has been closed, and the only resource seems to be the Wiki page at https://github.com/ct-Open-Source/tuya-convert/wiki/Collaboration-document-for-PSK-Identity-02

But it feels a bit silly to make random suggestions in that Wiki page, or should I do so? Or is there a discussion forum somewhere where I could drop suggestions?

One suggestion I have: I've opened up a freshly purchased lightbulb, so I now have the virgin TYLC6E module sitting on my desk. I could get another one from the same store and repeat the process, then I could read out the flash of both modules and possibly rule out (or confirm, which would be unfortunate) a factory-installed unique PSK seed.

Another suggestion is that we could try flashing the image onto an ESP8266 module with JTAG support (which I think exist nowadays) so we could run the initial contact of the module under a debugger.

doofy, 2020-11-23:
I would join you. See #820

AlexGustafsson, 2020-11-25:
I'm interested in helping if possible. I've intercepted network traffic for the entire configuration and setup of a brand new Deltaco SH-P01, both from the device to the cloud, between the device and phone as well as the phone and cloud (Deltaco Smart Home App). It seems to me that the device was already pre-configured with the new firmware, so I don't think I've been enable to intercept a firmware update (which seems to be wanted, judging from the discussion on #483).

I'm a bit reluctant to share the raw captures for what I assume are obvious reasons, but is there anything I should look for?

doofy, 2021-01-25:
See #834

PalasX, 2021-03-20:
I just bought a 4 pack of Jinvoo SM-PW701U Plug, all with the new PSK and unflashable. Are any of the primary contributors looking for donor equipment to test?