ct-Open-Source/tuya-convert #828: finally successful, why did it not flash with an Android Smartphone, but with a Windows Laptop connected?

r0tati0n, 2020-11-23:
I had loads of trouble flashing gosund SP1 plugs. I followed the wiki, connected an Android device, it did not work.
Then I used a Windows 10 laptop to connect instead of the Android smartphone, Firefox on the laptop automatically opened a greeter page telling me that I am connected to Wifi, and suddenly flashing worked.

Why did flashing not work with the Android Device?
What is the difference here?
Ist it because the greeter page did not open?
Should I have gone to a page without https (neverssl.com) to get the greeterpage and then it would work?

majki09, 2020-11-28:
Same here. I was getting
```
...................
SmartConfig complete.
Resending SmartConfig Packets
```
all the time. But when I saw this issue and used a laptop with Windows instead of Android smartphone the script was able to continue flashing:
```
Stopping smart config
Fetching firmware backup
```

drumstick77, 2020-12-01:
Hi, I'm getting the same problem. I've got the tuya convert installed on pi4, and using the wemos d1 mini as donor and getting the same problem.

when you guys said to use windows instead of android smartphone, do you mean that I use a windows laptop to connect to the vtrust instead of using d1 mini donor?

r0tati0n, 2020-12-02:
I used a panasonic laptop with debian for installing and creating the WIFI and a Windows 10 laptop for connecting to WLAN

majki09, 2020-12-02:
@drumstick77 When I got this message:

```
IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue
```
I've connected my Windows laptop to **vtrust-flash** over WiFi.

drumstick77, 2020-12-02:
Thanks. Do you have an instruction on creating WiFi on debian?

r0tati0n, 2020-12-02:
what for? it is done automatically via the script.

drumstick77, 2020-12-02:
Sorry, I probably got the wrong end of the stick here.
So are you saying that you installed tuya convert on a debian laptop instead of pi3/pi4?
If yes which debian flavour or image did you use? Do you have a link to it? Thanks

r0tati0n, 2020-12-02:
yes, did not use a PI. I use parrot OS.

drumstick77, 2020-12-02:
I am in UK, and I tried my method on aoycocr plugs and avatar plugs, they all are getting the same problem. I did 8bof aoycocr plugs 1.5 years with no issues. I'm wondering if thrive updated their hardware to disable this ability.