ct-Open-Source/tuya-convert #829: Docker help ?

tablatronix, 2020-11-25:
Ok, I am not very skilled in docker but this throwing me for a loop

Followed the instructions, 
What is up with the container name and service?

[image]

using debian, image

it runs ok with run, but then logging and interacting with the container to debug the AP issue , and I have no logging

tablatronix, 2020-11-25:
I tried adding a static container name to the config yml, same thing..

tablatronix, 2020-11-25:
ohh Do i need to run this in the docker folder after build?

tablatronix, 2020-11-26:
It works now, I built against 
`FROM phusion/baseimage:latest-arm`

Now I have to fix the dnsmasq issue

closed 2020-11-26 by tablatronix

tablatronix, 2020-11-26:
Finally got it to work!
never did get the screen logging working shrug..

it looks like the script needs to have some better start ap checking, I found out that the dns and ap were actually working , but crashing, sometimes it would show up for a bit but crash when I connected to it. I eventually got it to stay up by killing dnsmasq and hostapd that seemed to be stuck running. Ill look into it further to see what actually fixed it. Running start ap outside of docker seemed to help me get some better tests

[image]

kueblc, 2020-12-02:
Hi @tablatronix, glad you have figured it out. The docker configuration was submitted by the community. If you have any specific improvements to contribute, feel free to open a PR.