ct-Open-Source/tuya-convert #830: Merkury MI-BW904-999W, Merkury Innovations A21 Smart Light Bulb, 75W Color LED, no longer working

5ft24dave, 2020-11-27:
Bought 8 of these at Walmart.  On sale for 4.97 each.
Put the bulb in pairing mode (rapid flashing)  have tried with phone connected to the vtrust-flash network and the laptop connected.  neither made a difference.
gives the following.  after the first 2-3 dots after it sends the token and secret, the light goes full brightness and solid on like it has connected, but then it just keeps retrying over and over.  finally says Device did not appear with the intermediate firmware.

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................

Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N] n
======================================================
Cleaning up...
Closing AP
Exiting...

My feeling is Tuya patched the firmware in these before they left the factory.  These were virgin, never used any app on them.  straight out of the box.
Thanks

Any way I can get more info from the bulbs to help?

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5606161/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5606162/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/5606163/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/5606164/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5606165/smarthack-wifi.log)

kueblc, 2020-11-27:
Unfortunately these bulbs no longer use the ESP82xx so you will not be able to install ESP firmware like Tasmota.

closed 2020-11-27 by kueblc

5ft24dave, 2020-11-27:
decided to use the Geenie app and pair a bulb to see if I could get any info.  bulb shows main module firmware as 1.0.10 and MCU module as the same

5ft24dave, 2020-11-27:
ah.. ok.  Thanks

kueblc, 2020-11-27:
I was pretty bummed about this too, I saw the same sale.

5ft24dave, 2020-11-27:
Guess I could pair them with the Tuya app and just use the Tuya plugin for now... May disassemble one to see what they use.
Thanks

5ft24dave, 2020-11-27:
Colin,
Dissected one.  It is a proprietary module Tuya makes.

WB2L is a low-power embedded Wi-Fi module that Tuya has developed. It
consists of a
highly integrated RF chip (BK7231T) and several peripheral components, with
an
embedded Wi-Fi network protocol stack and robust library functions. WB2L is
embedded
with a low-power Arm Cortex-M4 MCU, 2 MB flash memory, and 256 KB static
random-access memory (SRAM), and has extensive peripherals.
WB2L is an RTOS platform that integrates all function libraries of the
Wi-Fi MAC and
TCP/IP protocols. You can develop embedded Wi-Fi products as required.

https://developer.tuya.com/en/docs/iot/device-development/module/wifibt-dual-mode-module/wb2l-datasheet?id=K9duegc9bualu

kueblc, 2020-11-27:
Yes I'm familiar, unfortunately I'm not aware of any mature alternative firmware for the BK7231T. Even if one was available, we'd need to find a new exploit to do OTA flashing.

Let me know if you come across anything though, who knows, it might not be a lost cause after all.