ct-Open-Source/tuya-convert #832: Globe Suite Smart Plug turns off when at "Sending secret"

3377VH, 2020-11-28:
I just purchase a pack of 3 Globe Suite smart plugs model# 50000, $20 at Canadian Tire, I thought it was a good deal.

I tried to flash one using the Tuya-Convert script, but the plug seems to turn itself off at the "Sending secret" level.
I also tried the Globe Suite software and it was able to find it...until I get to the "Sending secret" part. I did not set it up with the proprietary app by fear of getting its firmware updated automatically.
This is the first time I use tuya-convert, so maybe my installation is not correct.
I use a Linksys WUSB600N USB wifi adapter, on Ubuntu 20. My seconf wifi connection is either an iPhone or a Windows 10 computer. I tried different approach, but nothing seems to change. I followed the instruction from Globe to set the plug in EZ mode. 
Is there something I missed or did wrong?
Thanks in advance, here are my logs.

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5610855/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5610856/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/5610857/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/5610858/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5610859/smarthack-wifi.log)

kueblc, 2020-12-02:
I do not see any connections in your wifi log, but you report that one of the plugs did exit EZ config mode. Do you have other wifi logs available?

3377VH, 2020-12-03:
Yeah, no my plug won't connect while in EZ config mode, I tried all 3 and they all react the same. (I tried the fast and the slow blinking modes). I tried waiting a few minutes before step 3 **Press ENTER to continue**. I have no issues connecting other devices, I tried both an iPhone and a Windows machine:
    Using interface wlx00259c0a755b with hwaddr 00:25:9c:0a:75:5b and ssid "vtrust-flash"
    wlx00259c0a755b: interface state UNINITIALIZED->ENABLED
    wlx00259c0a755b: AP-ENABLED 
    wlx00259c0a755b: AP-STA-CONNECTED ac:7b:a1:aa:b3:51 **(Win10 computer)**
    wlx00259c0a755b: AP-STA-CONNECTED ee:c9:7b:c8:57:2a **(iPhone)**
    wlx00259c0a755b: AP-STA-DISCONNECTED ee:c9:7b:c8:57:2a

And as soon as I start the script, it goes:
    ====
    Starting smart config pairing procedure
    Waiting for the device to install the intermediate firmware
    Put device in EZ config mode (blinking fast)
    Sending SSID                  vtrust-flash
    Sending wifiPassword          
    Sending token                 00000000
    Sending secret                0101
    **.> here the plug exits EZ mode and turns back to previous (ON/OFF) state <...............**
    SmartConfig complete.
    Resending SmartConfig Packets
    ...
    ====
I tried changing the channel in the hostapd.conf and the region code in the main.py script (I'm in Canada), I get always the same result.

I setup one plug via the Tuyasmart app to see the firmware: Main and MCU modules are V1.0.7

Any suggestion?

dereitz, 2020-12-04:
I've run into the same situation with some recently purchased BN-Link BNC-60/U133TJ devices.  Would appreciate any suggestions as well.  Thanks!!

BobMoron, 2020-12-05:
I have the same issue with globe smart plugs (50201) as well as bulbs (34218 and 34202)

3gyptian, 2020-12-06:
@3377VH  .. yeah I just picked up the same deal at canadian tire and had same flashing issue.  Opened one up just now and it has the newer WB2S module in it unfortunately so these can't be converted ;(   

I've got another single Globe plug from London Drugs a few weeks ago that does have an expresif mac address so maybe I'll try that and if it works I'll pick up a few of those that might be around town.  

I'm afraid the days having ESP8266 modules in these smartplugs is coming to an end.  It was great while it lasted.

rpavlik, 2020-12-07:
I suspect this may be the same issue as the one I filed (#848 ) on some Costcto-sold Feit PLUGWIFIG2, as well as #846 which looks similar but might have an actual connection in the wifi log.

3377VH, 2020-12-07:
Thanks @3gyptian! That's too bad, I just started playing with HA, I'll have to be more careful when I choose my modules. I guess we can close this thread now. Thanks again!

closed 2020-12-07 by 3377VH

mondjef, 2020-12-26:
I too got the canadian tire deal and could not flash them.  I was able to successfully still use them in my OpenHab integration by using TuyApi and mqtt, however shortly getting everything set up I was then experiencing the random on/off cycles.  I decided to reach out to Globe to see what sort of support I would get and for the most part just wanted to rant a bit to them as I do like the plugs and the fact that they are 15A rated.  I can certainly say I was fully satisfied with the level of support that they gave me and their response times were very good.  At first they were just going to send out replacements for all 6 units I was having issues with, all they requested was the device ids for the units.  A couple days later they reached out again and their engineers had been working on a fix and were hoping to try to push an update to my units to test if resolve the issues...however after testing they contacted me again to inform me that they were not going to be able to push the upate after all for some reason and instead sent out replacement units to me promptly.  I just got replacement units a couplde of days ago and paired one device with the app and noticed it was using an older firmware and so I tried flashing and to my delight I was able to flash tasmota to it.  Tried the other 5 units they sent me and same thing....they are all using tasmota now and I couldn't be happier.  Maybe they got some old stock still kicking around so my advice if anyone is experiencing the random switching issues and want to get it resolved and be able to possibly load tasmota to reach out to Globe for support.  In any event, if they send you units that are no longer defective but can't be converted you can still use locally via the TuyApi integration so I think it is not a complete lost cause.

3gyptian, 2021-03-15:
They were cheap so I ended up buying another 3 pack of the same Globe Smart switches and although they did have ESP8266's the firmware was too new to convert:

```
Your device's firmware is too new.
Tuya patched the PSK vulnerability that we use to establish a connection.
You might still be able to flash this device over serial.
For more information and to follow progress on solving this issue see:
https://github.com/ct-Open-Source/tuya-convert/wiki/Collaboration-document-for-PSK-Identity-02
```
So I ended up flashing via serial. They're easy to open with a small sharp screwdriver banged into one of the bottom edges and worked around .. pops open easy.  

Then configured in esphome as per: https://templates.blakadder.com/globe_smart.html

Where GPIO's used are:
GPIO04	Led1i
GPIO12	Relay1
GPIO14	Button1

Good times ..