ct-Open-Source/tuya-convert #834: Bug bounty for new PSK?

imduffy15, 2020-11-30:
Hi All,

Is there a bug bounty we can donate to in order to put some crash behind working around the new PSK encryption?

Thanks,
Ian

bani6809, 2020-12-03:
new PSK is worrying, 80% of the devices i've purchased recently have switched to it or abandoned esp entirely and switched to wb2l. the pool of hackable devices is disappearing quickly.

aSmig, 2020-12-05:
https://gitcoin.co/ would probably be a good platform for putting some money into this.

abvgeej, 2020-12-15:
@aSmig BountySource is, imho, a more established platform
https://www.bountysource.com/issues/86360018-new-psk-format

play113, 2021-01-23:
ich würde mich gern kostentechnisch daran beiteiligen um meine
"Neo Coolcam Smart Wifi-Plug-16A" von ihren chinesischen freundesbanden zu befreien ...

habe das gleiche problem:
new client on port 443 from 10.42.42.14:12282
ID: 0242416f68626d643661473931494652319a30f4e306cca0221796ddcf6b991d50fac6583162ee569a8054aae3fb3b384f
PSK: 1a4b4792de50e7463b29afa89fd6f9d57fbb77193f55d94dd46dfd2c49d11c2f
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)