ct-Open-Source/tuya-convert #835: Unable to Flash Lumiman LM530 - E26

rgeyer, 2020-11-30:
**Are you reporting a successful flash?**
No

**Do you have a question about configuring your newly flashed device?**
No

**Did you have issues getting a device to flash?**
Yes..

I'm trying to flash a Lumiman LM530-E26 RGB smart bulb. It works with "SmartLife" and appears to be a Tuya device. I *did* see in the wiki that "Success?" is marked as NO for this device, but also saw several reviews and Reddit messages indicating that they had success flashing this device.

I tried this both on a Raspberry Pi 3B+ and my Ubuntu laptop.

On the pi, there it seemed to loop on disconnecting from wifi, so I tried with Ubuntu. The screenlog.0.txt attached is from the Ubuntu attempt(s).

d4:a6:51:22:ad:c0 is the device
7c:d1:c3:6b:d8:44 is a Macbook I'm using to stay connected to the vtrust-flash SSID.

Not sure if these are simply not supported, or if I'm having a string of bad luck. I tried two devices, without success.

[screenlog.0.txt](https://github.com/ct-Open-Source/tuya-convert/files/5613462/screenlog.0.txt)

kueblc, 2020-12-02:
Unfortunately `ac:7b:a1:aa:b3:51` is not an ESP82xx based device. We can only flash ESP firmware on this type of microcontroller. Manufacturers have been slowly and silently switching to other microcontrollers, for cost or "security".

closed 2020-12-02 by kueblc

rgeyer, 2020-12-02:
Thank you @kueblc. Tho... Where did the MAC `ac:7b:a1:aa:b3:51` come from? It does not appear in my logs. Perhaps you had a different MAC in your clipboard when you posted?

In any case, I understand that manufacturers have been changing the microcontrollers, and that my device may no longer be supported. Sad, but inevitable. Just wanted to make sure we're talking about the same device.

Thanks again!

kueblc, 2020-12-02:
Good catch @rgeyer, that's exactly what happened, bad copy paste. I double checked with `d4:a6:51:22:ad:c0` and it is unfortunately still not an ESP82xx.

You may be able to use tuyapi to control the device locally, and block cloud access with your local network's DNS functions.

rgeyer, 2020-12-02:
Thanks again @kueblc I'll look into it. 'Preciate all of the efforts and support!