ct-Open-Source/tuya-convert #836: install_prereq.sh: problem with dnsmasq

jcisio, 2020-11-30:
Hello,

I try to run this tool on my PC (Ubuntu 20.04) but I have a problem in the first step, because the is another DNS server listenning on port 53, so that dnsmasq cannot be installed.

Is it required to install dnsmasq? Could you tell me what does it do? For the rest, and mainly hostapd I can guess what do they do. What should I configure manually my DNS (or modify my host file) instead of installing dnsmasq?

imduffy15, 2020-11-30:
It serves DHCP for 10.42.42.0/24 and causes all DNS requests to resolve to 10.42.42.1 so all the tuya cloud requests get converted to the stuff your running locally.

jcisio, 2020-11-30:
Thank you. I guess I'll run this script in a container with a previous version of Ubuntu instead of inspecting the DNS requests to Tuya cloud.

jcisio, 2020-11-30:
I'll leave this issue open. I think it could help other if it says something and exits when it detectes Ubuntu 20.04 or later, because you have to uninstall systemd-resolve before installing dnsmasq.

jcisio, 2020-12-02:
It turns out that it is not a problem.

closed 2020-12-02 by jcisio