ct-Open-Source/tuya-convert #837: Can't stay connected to AP, can't flash device, possible new PSK situation?

pullaf, 2020-11-30:
As per title, I've been trying to flash my new smart power strip for a couple days with no success
At first I tried with Ubuntu running on a laptop, can't create AP
Then an Ubuntu VM, can't see the created AP for some reason, cheap wifi dongle might not have proper driver support
Finally with a Rpi4, can't stay connected to vtrust-flash so I decide to give the old Rpi3 a go as well, same problem
My phone connects to the network and then drops out of it right away, running Android 11, the W10 PC connects but then fails to open the portal page, not sure if it stays connected
I tried all sort of fixes found in here, from running hostapd as standalone (phone will connect but won't get an IP so it disconnects after a bit) to unblocking everything, still no luck
So anyway, I was going to blame it on the AP but then I checked the logs and turns out the -psk one has some content, I'll paste part of it (omitting the repeated parts)

`new client on port 443 from 10.42.42.37:16902
ID: 0242416f68626d6436614739314946523144dab89cc54b24ca91d1344de520f5a8f494134099c51bd37154e7edb41a4a85
PSK: 30630c212d110e5bb225c2049541ed9cff1f676cd6a990a1ae0a75c92607dbd1
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.37:47947
ID: 0242416f68626d6436614739314946523144dab89cc54b24ca91d1344de520f5a8f494134099c51bd37154e7edb41a4a85
PSK: 30630c212d110e5bb225c2049541ed9cff1f676cd6a990a1ae0a75c92607dbd1
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.37:58303
ID: 0242416f68626d6436614739314946523144dab89cc54b24ca91d1344de520f5a8f494134099c51bd37154e7edb41a4a85
PSK: 30630c212d110e5bb225c2049541ed9cff1f676cd6a990a1ae0a75c92607dbd1
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.37:30318
ID: 0242416f68626d6436614739314946523144dab89cc54b24ca91d1344de520f5a8f494134099c51bd37154e7edb41a4a85
PSK: 30630c212d110e5bb225c2049541ed9cff1f676cd6a990a1ae0a75c92607dbd1
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
^CTraceback (most recent call last):
  File "./psk-frontend.py", line 113, in <module>
    main()
  File "./psk-frontend.py", line 106, in main
    r,_,_ =  select.select(readables, [], [])
KeyboardInterrupt
`

The wifi log on the other hand it just this over and over

`System info
===========
94acc3b
Linux raspberrypi 5.4.72-v7+ #1356 SMP Thu Oct 22 13:56:54 BST 2020 armv7l GNU/Linux
OpenSSL 1.1.1d  10 Sep 2019
Dnsmasq version 2.80  Copyright (c) 2000-2018 Simon Kelley
Compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile

This software comes with ABSOLUTELY NO WARRANTY.
Dnsmasq is free software, and you are welcome to redistribute it
under the terms of the GNU General Public License, version 2 or 3.
hostapd v2.8-devel
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2019, Jouni Malinen <j@w1.fi> and contributors
Python 3.7.3
===========
Configuring AP interface...
RTNETLINK answers: File exists
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: ./hostapd.conf
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:59:74:4d and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
wlan0: AP-STA-CONNECTED c4:4f:33:bf:90:40
wlan0: AP-STA-CONNECTED 00:02:6f:8e:05:08
wlan0: AP-STA-DISCONNECTED c4:4f:33:bf:90:40
wlan0: AP-STA-CONNECTED c4:4f:33:bf:90:40
wlan0: AP-STA-DISCONNECTED c4:4f:33:bf:90:40
wlan0: AP-STA-CONNECTED c4:4f:33:bf:90:40
wlan0: AP-STA-DISCONNECTED 00:02:6f:8e:05:08
wlan0: interface state ENABLED->DISABLED
wlan0: AP-STA-DISCONNECTED c4:4f:33:bf:90:40
wlan0: AP-DISABLED 
wlan0: CTRL-EVENT-TERMINATING 
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...`

Let me know if I can be of any help with the new PSK, as long as someone can guide me through the soldering the right pins part

[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5617044/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/5617045/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/5617046/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5617047/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5617048/smarthack-mqtt.log)

fzimmermann89, 2020-12-01:
... ``ID: 02...`` 
should mean new PSK and not supported

pullaf, 2020-12-01:
So the phone not staying connected doesn't play any part in this?
Is there an ETA on the new PSK cracking? I'm still a beginner at soldering so I'd rather wait a bit and do it the tuya-convert way, but if it's still nowhere to be seen AND I could help by dumping the current firmware I might just give it a try

daniel-lo, 2020-12-02:
I don't know if my problem is related. My Phones (Xiaomi/Nokia) both don't stay connected to the AP either. It tells me it's connected but no internet .. connection drops .. and then this starts over.

Is this the way it is supposed to be?

kueblc, 2020-12-02:
@pullaf @fzimmermann89 is correct, PSK ID beginning in 02 is unfortunately not supported. There is no way to know when or if we will find another exploit.

@daniel-lo there is no internet provided by `vtrust-flash`. It seems that some phones refuse to stay connected to a network without internet unless you explicitly select to stay connected anyway.

closed 2020-12-02 by kueblc